analogdevicesinc/aditof_sdk #41: Output directories of demo software in Windows

yhanalog, 2019-06-14:
According to the the user guide, the application binaries for the example demo software are supposed to be generated in the 'aditof_sdk/build/example/aditof-demo/Release' directory'.

However, binaries are found in the following directory,

aditof_sdk\scripts\build\examples\aditof-demo\Release\aditof-demo.exe
aditof_sdk\scripts\build\sdk\Release\aditof.dll

Furthermore, opencv dll had to be copied to the corresponding directory manually to run the demo.

dNechita, 2019-06-14:
Hi,

Regarding the directory issue: 
It matters from what location you run the script (generate_msvc2015_solution.bat). The script creates a build folder to the same location where you run it from.  It looks like you cd to aditof_sdk\scripts and ran the script from there. The build can be generate anywhere, it's a user preference.

Regarding the need to copy the opencv dll:
We've recently updated the documentation and added a few more instructions for the opencv installation. You need to define two opencv environment variable. This ensures that no dll copying is required. If this doesn't work for you, please let me know.

Thanks for taking the time to report any inconsistencies or issues.

yhanalog, 2019-06-14:
Thank you very much for prompt response.I
- Adding the path (in addition to two environment variables) for opencv eliminated the need of open cv dll copying.
 
- The Visual Studio solution generates aditof.dll and aditof-demo.exe into different places, regardless of where I run the script, so one of them needs to be moved to run the aditof-demo.exe. The demo software runs OK after copying the aditof.dll into the target directory, so it is not a big issue. I used Visual Studio 2017 by replacing "Visual Studio 14 2015" with "Visual Studio 15 2017". 

Thank you again for your help.

dNechita, 2019-06-19:
Glad to hear that things work.
Indeed, aditof.dll needs to be copied next to aditof-demo.exe. To make things easier, you can add a custom step in Visual Studio so that each time you perform a build, the aditof.dll gets copied next to the demo executable.

closed 2019-06-19 by dNechita