analogdevicesinc/aditof_sdk #54: Powerup Dragon Board /w linux Causes Reboot

bitslip6, 2019-07-19:
Hi, I have the development kit with dragon board connected to a linux system running on ARM (Raspberry Pi3B+) .  When I power up the dragon board and it is connected via usb to the Pi the DS4 LED lights briefly followed by an immediate reboot of the dragon board.

When the system is connected to a windows system it works just fine.

Any help on debugging this issue wold be most appreciated!

Thanks!

bitslip6, 2019-07-24:
> Hi, are you powering the RasbberryPi separately? (for example with: https://www.raspberrypi.org/products/raspberry-pi-universal-power-supply)

Yes all systems have their own power supply.   Do we need common ground?

dNechita, 2019-07-25:
I'm not sure what is going on. The DS4 LED should light only when S3 button is pressed provided that the OS on the Dragonboard has finished booting. Does you led light up (even briefly) automatically?

santosh-madanwad, 2019-08-01:
Sorry to jump in on this discussion, but I recently tried this same scenario and wanted to share my observations. 
I set up a RPi3B+ board with 'Ubuntu Mate' SD card image that I took from below location;-
    https://ubuntu-mate.org/download/
I connect Dragonboard (with 3DToF Mezzanine) to RPi over USB and then press 'S3' button. I see DS4 light up, and also see that RPi recognizes the TOF Camera correctly (I check the dmesg and see messages same as shown [Linux User Guide](https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/linux/user_guide.md)

@bitslip6 If you can share OS details on your RPi3B+, I might be able to help.

closed 2019-10-18 by dNechita