analogdevicesinc/aditof_sdk #61: Raspberry PI3: eeprom permission issue

bitslip6, 2019-08-13:
Getting permission denied error writing to the eeprom in running the demo app.  It's unable to open  /sys/bus/i2c/devices/1-0056/eeprom, 1-0056 is a symlink to:
/sys/devices/platform/soc/3f804000.i2c/i2c-1/1-0056/

permissions on i2c sys files:
`pi@raspberrypi:~/workspace/github/aditof_sdk/build/examples/aditof-demo $ ls -dls /sys/devices/platform/soc/3f804000.i2c/i2c-1/1-0056/
0 drwxr-xr-x 3 root root 0 Jul 10 17:01 /sys/devices/platform/soc/3f804000.i2c/i2c-1/1-0056/`

and here is the relevant strace

`stat64("/dev/video0", {st_mode=S_IFCHR|0660, st_rdev=makedev(0x51, 0x3), ...}) = 0
openat(AT_FDCWD, "/dev/video0", O_RDWR|O_NONBLOCK) = 10
ioctl(10, VIDIOC_QUERYCAP, {driver="unicam", card="unicam", bus_info="platform:unicam 3f801000.csi", version=4.19.57, capabilities=V4L2_CAP_VIDEO_CAPTURE|V4L2_CAP_READWRITE|V4L2_CAP_STREAMING|V4L2_CAP_DEVICE_CAPS|V4L2_CAP_EXT_PIX_FORMAT, device_caps=V4L2_CAP_VIDEO_CAPTURE|V4L2_CAP_READWRITE|V4L2_CAP_STREAMING|V4L2_CAP_EXT_PIX_FORMAT}) = 0
stat64("/dev/video0", {st_mode=S_IFCHR|0660, st_rdev=makedev(0x51, 0x3), ...}) = 0
openat(AT_FDCWD, "/dev/video0", O_RDWR|O_NONBLOCK) = 11
openat(AT_FDCWD, "/sys/bus/i2c/devices/1-0056/eeprom", O_RDWR|O_CREAT|O_TRUNC, 0666) = -1 EACCES (Permission denied)
write(2, "Error eeprom_open: Permission de"..., 37) = 37
gettid()                                = 1215
getpid()                                = 1215
uname({sysname="Linux", nodename="raspberrypi", ...}) = 0
openat(AT_FDCWD, "/tmp//aditof-demo.raspberrypi.root.log.WARNING.20190710-165806.1215", O_WRONLY|O_CREAT|O_EXCL, 0664) = 12
fcntl64(12, F_SETFD, FD_CLOEXEC)        = 0
fcntl64(12, F_GETFL)                    = 0x1 (flags O_WRONLY)
fcntl64(12, F_SETFL, O_WRONLY|O_APPEND) = 0
_llseek(12, 0, [0], SEEK_END)           = 0
unlink("/tmp//aditof-demo.WARNING")     = 0
symlink("aditof-demo.raspberrypi.root.log.WARNING.20190710-165806.1215", "/tmp//aditof-demo.WARNING") = 0
fstat64(12, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
write(12, "Log file created at: 2019/07/10 "..., 209) = 209
getpid()                                = 1215
write(2, "W0710 16:58:06.696756  1215 loca"..., 68) = 68
openat(AT_FDCWD, "/sys/bus/i2c/devices/1-0056/eeprom", O_RDWR|O_CREAT|O_TRUNC, 0666) = -1 EACCES (Permission denied)
write(2, "Error eeprom_open: Permission de"..., 37) = 37
`

quick background.  I have verified r98 and R109 shorts and JP1 short of pin 2&3.  (I removed the short on JP1 between pin 2&3, however my multi-meter shows connection between pad 1 and 2?)

I also verified connectivity between all of these pins on pi and aditof board
Pi pin 3 is split and connected to ADITOF pins 15 and 19
Pi pin 5 is split and connected to ADITOF pins 17 and 21

Any help would be appreciated!

bitslip6, 2019-08-13:
Fixed.  I2C connection issue.

closed 2019-08-13 by bitslip6

Sebbbastian, 2020-06-16:
How did you actually fix this issue?

bitslip6, 2020-06-16:
Can't quite remember.  maybe I was missing ground connection between pi and AFE ?