analogdevicesinc/aditof_sdk #62: Raspberry Pi: getFrame() timeout

bitslip6, 2019-08-13:
Unable to play video feedback.  seems to be timeout on local_device.cpp:467
r = select(m_implData->fd + 1, &fds, NULL, NULL, &tv);

I'm a bit stumped as to how to debug this further.  It appears the eeprom is working since we were able to read calibration data.  Any help on debugging this would be very appreciated.

here is the strace output:
```
write(2, "I0813 23:51:36.186231  1662 cali"..., 76) = 76
mmap2(NULL, 483328, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x65b61000
poll([{fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 2, 0) = 1 ([{fd=13, revents=POLLIN}])
read(13, "\3\0\0\0\0\0\0\0", 16)        = 8
futex(0x71e1e980, FUTEX_WAKE_PRIVATE, 2147483647) = 0
poll([{fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 2, 0) = 0 (Timeout)
poll([{fd=12, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=12, revents=POLLOUT}])
writev(12, [{iov_base="\23\0\3\0\3\0@\1\272\1\0\0\f\1\4\0\3\0@\1@\0\0\0\0\0\0\0\22\0\17\0"..., iov_len=328}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 328
poll([{fd=12, events=POLLIN}], 1, -1)   = 1 ([{fd=12, revents=POLLIN}])
recvmsg(12, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\34\0\322\0\3\0@\1#\0\0\0\374\317J\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 192
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 2, 3) = 0 (Timeout)
mmap2(NULL, 483328, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x65aeb000
munmap(0x65bd7000, 483328)              = 0
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 2, 10) = 1 ([{fd=12, revents=POLLIN}])
recvmsg(12, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\34\0\332\0\3\0@\1V\1\0\0\24\320J\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
brk(0x2291000)                          = 0x2291000
munmap(0x65aeb000, 483328)              = 0
recvmsg(12, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\34\0\332\0\3\0@\1p\1\0\0\26\320J\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 320
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=12, events=POLLIN}, {fd=13, events=POLLIN}], 2, 0) = 0 (Timeout)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(12, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
```

and the log output:
```
I0814 00:02:16.056355  2087 system_impl.cpp:31] System initialized
I0814 00:02:16.057685  2087 camera_impl.cpp:19] Initializing camera
I0814 00:02:17.021828  2087 calibration.cpp:224] EEPROM calibration data size 65044
I0814 00:02:25.665418  2087 camera_impl.cpp:34] Camera initialized
I0814 00:02:25.670692  2087 local_device.cpp:203] Starting device
I0814 00:02:25.984545  2087 camera_impl.cpp:54] Chosen mode: near
I0814 00:02:25.995613  2087 calibration.cpp:104] Found firmware for mode near
I0814 00:02:25.997917  2087 calibration.cpp:110] Firmware size: 5220
I0814 00:02:26.000075  2087 camera_impl.cpp:85] Firmware size: 5220
I0814 00:02:27.317703  2087 calibration.cpp:151] Calibration gain: 1
I0814 00:02:27.317924  2087 calibration.cpp:155] Calibration offset: 0
I0814 00:02:27.334374  2087 calibration.cpp:151] Calibration gain: 1
I0814 00:02:27.334597  2087 calibration.cpp:155] Calibration offset: 0
I0814 00:02:27.354959  2087 calibration.cpp:151] Calibration gain: 1
I0814 00:02:27.355227  2087 calibration.cpp:155] Calibration offset: 390.23
I0814 00:02:27.376343  2087 calibration.cpp:151] Calibration gain: 1.22014
I0814 00:02:27.376565  2087 calibration.cpp:155] Calibration offset: 2003.8
W0814 00:02:36.445209  2126 local_device.cpp:474] select timeout
W0814 00:02:36.445317  2126 camera_impl.cpp:208] Failed to get frame from device
```

__

acozma, 2019-08-14:
Have you made all the changes to the board listed [here](https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/raspberrypi3/user_guide.md#modifying-the-ad-96tof1-ebz-revb-to-work-with-the-rpi)? 

If these are not made the communication with the board will not work properly.

bitslip6, 2019-08-14:
> Have you made all the changes to the board listed [here](https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/raspberrypi3/user_guide.md#modifying-the-ad-96tof1-ebz-revb-to-work-with-the-rpi)?
> 
> If these are not made the communication with the board will not work properly.

Yes, R98 and R109 are shorted and I removed the short between pins 1 and 2 on J1 and shorted pins 2 and 3.

acozma, 2019-08-14:
Ok, that's good. There's one thing that was missed in the documentation: switch S1 must be moved to the ON position, need to flip it from how it is when the system works with the DragonBoard

bitslip6, 2019-08-15:
With the switch toggled on I am now getting video out.  Thanks for this help!

closed 2019-08-15 by bitslip6