analogdevicesinc/aditof_sdk #151: can you provide calibration tool for depth value？

KirkKong, 2019-11-01:
can you provide calibration tool or method for depth value？

dNechita, 2019-11-04:
The value of each element of a frame is depth. The unit is mm.

KirkKong, 2019-11-07:
> The value of each element of a frame is depth. The unit is mm.

@dNechita
Thanks for your reply!
I mean how to create the calibration data stored in EEPROM, such as Gain and Offset.
Do you have user guide or calibration tool?

acozma, 2019-11-07:
The calibration software and user guide is currently in the final development stage and it will be released in the next few weeks.

KirkKong, 2019-11-07:
> The calibration software and user guide is currently in the final development stage and it will be released in the next few weeks.

@acozma 
Thanks a lot!
I will pay attention to the changes in the next few weeks.

dNechita, 2020-05-04:
The calibration software is available.
Doc is at: https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/calibration

closed 2020-05-04 by dNechita