analogdevicesinc/aditof_sdk #152: Trouble building on Raspberry Pi

bitslip6, 2019-11-04:
Hi, first I would like to tank you for all of the hard work on this SDK!

Now our issue.

We are having difficulty building the latest source on raspberry pi.  We are using the scripts/raspberrypi/setup.sh compilation script.  There are several errors with the current script.  

First, it does not seem to download opencv-3.4.1.zip correctly.  I had to manually download from opencv, unzip and compile. This got us a bit further.

Second, protobuf does not install correctly with the current scripts.  The source for 3.9.0 is downloaded correctly, however it was not compiled or installed and opencv (via cmake) was unable to find it.  I followed the build procedure and compiled in deps/protobuf (./autogen.sh; ./configure; make; make install) and then  opencv cmake (autoconf) was able to find protobuf 3.9.0 correctly in /usr/local/.

After that the build seems to complete successfully with some warnings on the python3 bindings for opencv.  Here are the last lines from the setup.sh script:

/```
home/pi/workspace/github/aditof_sdk/deps/opencv-3.4.1/build_3_4_1/modules/python_bindings_generator/pyopencv_generated_ns_reg.h:2055:35: warning: cast between incompatible function types from ‘PyObject* (*)(PyObject*, PyObject*, PyObject*)’ {aka ‘_object* (*)(_object*, _object*, _object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
     {"setUseOpenCL", (PyCFunction)pyopencv_cv_ocl_setUseOpenCL, METH_VARARGS | METH_KEYWORDS, "setUseOpenCL(flag) -> None\n."},
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/workspace/github/aditof_sdk/deps/opencv-3.4.1/build_3_4_1/modules/python_bindings_generator/pyopencv_generated_ns_reg.h:2056:32: warning: cast between incompatible function types from ‘PyObject* (*)(PyObject*, PyObject*, PyObject*)’ {aka ‘_object* (*)(_object*, _object*, _object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
     {"useOpenCL", (PyCFunction)pyopencv_cv_ocl_useOpenCL, METH_VARARGS | METH_KEYWORDS, "useOpenCL() -> retval\n."},
                                ^~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [modules/python3/CMakeFiles/opencv_python3.dir/build.make:63: modules/python3/CMakeFiles/opencv_python3.dir/__/src2/cv2.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7520: modules/python3/CMakeFiles/opencv_python3.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
/home/pi/workspace/github/aditof_sdk
/home/pi/workspace/github/aditof_sdk/build /home/pi/workspace/github/aditof_sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/workspace/github/aditof_sdk/build
[ 74%] Built target aditof
[ 85%] Built target first-frame
[ 85%] Built target first-frame-ethernet
[100%] Built target aditof-demo
```

Then running first frame gives what looks like an I2C communication error:
```
pi@raspberrypi:~/workspace/github/aditof_sdk $ sudo ./build/examples/first-frame/first-frame 
I1104 19:24:18.819532  3381 device_enumerator_raspberry.cpp:16] Looking for devices on the target
I1104 19:24:18.824537  3381 device_enumerator_raspberry.cpp:49] Looking at: /dev/video0 for an eligible TOF camera
I1104 19:24:18.826802  3381 system_impl.cpp:33] System initialized
I1104 19:24:18.826993  3381 camera_96tof1.cpp:20] Initializing camera
I1104 19:24:18.827055  3381 local_device.cpp:118] Opening device
Error eeprom_open: No such file or directory
W1104 19:24:18.835683  3381 local_device.cpp:199] EEPROM not available!
W1104 19:24:18.836746  3381 local_device.cpp:591] EEPROM not available!
W1104 19:24:18.837358  3381 local_device.cpp:572] EEPROM not available!
I1104 19:24:18.837551  3381 calibration.cpp:220] EEPROM calibration data size 100
W1104 19:24:18.837949  3381 local_device.cpp:572] EEPROM not available!
W1104 19:24:18.838145  3381 calibration.cpp:232] Failed to read from eeprom
W1104 19:24:18.838404  3381 camera_96tof1.cpp:31] Failed to read calibration data from eeprom
E1104 19:24:18.838639  3381 main.cpp:33] Could not initialize camera!
Segmentation fault
```

ldd output for first-frame:
```
pi@raspberrypi:~/workspace/github/aditof_sdk $ ldd ./build/examples/first-frame/first-frame 
	linux-vdso.so.1 (0x7ef2b000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f22000)
	libaditof.so.1.0 => /home/pi/workspace/github/aditof_sdk/build/sdk/libaditof.so.1.0 (0x76e1d000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76dcc000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76c85000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76c03000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76bd6000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76a88000)
	libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0x76a6f000)
	libprotobuf.so.20 => /usr/local/lib/libprotobuf.so.20 (0x7684b000)
	libwebsockets.so.14 => /home/pi/workspace/github/aditof_sdk/deps/installed/websockets/lib/libwebsockets.so.14 (0x76806000)
	libssl.so.1.1 => /usr/lib/arm-linux-gnueabihf/libssl.so.1.1 (0x76787000)
	libcrypto.so.1.1 => /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1 (0x76576000)
	/lib/ld-linux-armhf.so.3 (0x76f37000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x7654b000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76538000)
```

Finally i2cdetect seems to find the camera correctly:
```
pi@raspberrypi:~/workspace/github/aditof_sdk/deps/protobuf $ i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~/workspace/github/aditof_sdk/deps/protobuf $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

The setup works fine with the raspberry pi 3 image in github.  The system breaks after performing git pull and the above steps.  Any help in debugging this would be most appreciated!!

Thank you so much!

DanielGuramulta, 2019-11-05:
Hi,

I have started looking into your issue, as soon as I have an update I will let you know.

DanielGuramulta, 2019-11-05:
Hi,

1) Protobuf not building correctly with the setup.sh script on raspberry pi is due to a missing cxx flag. This pull-request should fix the issue #153 
2) OpenCV seems to have a bug on version 3.4.1 for raspberry pi, that is preventing it from building. I will work on a patch and an update for the script to make sure everything works fine when building opencv 3.4.1 using the setup.sh script. For now using `opencv 3.4.1 is only a dependency for the `opencv bindings`. The rest should work fine with only the version provided by `apt-get`.
3) What hardware revision are you using? If it is `revision B`, then when building with cmake, pass `-DWITH_REVB=1` for the raspberry pi build, everything should work like before. Sorry for not documenting this step, I will make sure to update the documentation accordingly.

bitslip6, 2019-11-19:
Hi.  I have everything compiling now.  I ended up just creating my own build scripts to work around the issues you mentioned.

I am now working on porting the SDK to the NXP mx8m mini: https://www.nxp.com/design/development-boards/i.mx-evaluation-and-development-boards/evaluation-kit-for-thebr-i.mx-8m-mini-applications-processor:8MMINILPD4-EVK

I can build the SDK on just about any platform now, however I'm having difficulty compiling a working linux kernel.  I have been trying to follow the guide here:

https://wiki.analog.com/resources/tools-software/linux-build/generic/raspberrypi

however I am unable to build a bootable kernel on a raspberry pi 4.  Where should I be looking to be able to get the addi9036 drivers building?  https://github.com/analogdevicesinc/linux/blob/adi-4.19.0/drivers/media/i2c/addi9036.c 

Your help is greatly appreciated!!

btogorean, 2019-11-19:
Hi,
If you want to port the camera driver on NXP platform I recommend you to try to apply the commit [e6942c](https://github.com/analogdevicesinc/linux/commit/e6942ce7c41d16d4070a3825f1b1e0491043fb8b)
on NXP 4.14 kernel tree. It was not tried by us because we don't support the NXP platform but i should work.

To build the RPi4 kernel you have to use adi_bcm2711_defconfig and take care that ADDI9036 driver will be build as module. So you have to copy it on SD card rootfs in /lib/modules/

bitslip6, 2019-11-20:
Thank you so much for this information.

Can you please  provide some detail on why this is a custom kernel and not just a loadable kernel module for the ADDI9036 driver?

m-arshad, 2019-11-22:
Hi,

I am also stuck with the same issue about missing protobuf. I have then seperatly compiled and installed the protobuf but now the make step. 
Below is what I am doing:

` pi@raspberrypi: /workspace/github/aditof_sdk/build $ cmake -DRASPBERRYPI=1 -DCMAKE_PREFIX_PATH="opt/glog;usr/local/lib;/opt/websockets" ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
/usr/local/bin/protoc: error while loading shared libraries: libprotobuf.so.20: cannot open shared object file: No such file or directory
CMake Warning at /usr/share/cmake-3.13/Modules/FindProtobuf.cmake:495 (message):
  Protobuf compiler version doesn't match library version 3.9.0
Call Stack (most recent call first):
  sdk/CMakeLists.txt:6 (find_package)

-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread (found suitable version "3.9.0", minimum required is "3.9.0") 
-- Found OpenCV: /usr (found version "3.2.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/workspace/github/aditof_sdk/build
pi@raspberrypi:/workspace/github/aditof_sdk/build $ make -j4
[  2%] Running cpp protocol buffer compiler on src/buffer.proto
/usr/local/bin/protoc: error while loading shared libraries: libprotobuf.so.20: cannot open shared object file: No such file or directory
make[2]: *** [sdk/CMakeFiles/aditof.dir/build.make:63: sdk/buffer.pb.h] Error 127
make[1]: *** [CMakeFiles/Makefile2:90: sdk/CMakeFiles/aditof.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`

DanielGuramulta, 2019-11-25:
`-DCMAKE_PREFIX_PATH="opt/glog;usr/local/lib;/opt/websockets"`, from what I see here, you installed protobuf in usr/local/lib? (or in usr/local/lib you have only the .so for protobuf, in this case the path should be /usr/local). Also make sure to have the correct path to `opt` which is `/opt`, same for `usr`
Try: `-DCMAKE_PREFIX_PATH="/opt/glog;/usr/local;/opt/websockets"` 
If this doesn't work you can use the script `setup.sh` available here: https://github.com/analogdevicesinc/aditof_sdk/tree/master/scripts/raspberrypi3.

DanielGuramulta, 2019-12-17:
@bitslip6 @m-arshad. Do you still have problems with building the SDK for the Raspberrypi platform? Please let me know!

Daniel.

closed 2020-01-09 by DanielGuramulta