analogdevicesinc/aditof_sdk #206: Register write doesn't work with the latest SD card image for 96 boards

yhanalog, 2020-02-06:
Hello,
We see an issue in writing any registers using low level API with the latest SD card image download from Github for last few weeks. Here is the summary
- An example that comes with the SDK was used for testing (aditof-demo)
- Code was compiled and executed with Visual Studio 2017 on Windows 10 machine.
- The camera system crashes if small signal removal is enabled. We initially observed this issue when we tried to write some other registers, and we see the same issue with the existing example software before modification.
- We happened to have an old SD card image which was taken few months ago and we don't see this problem. We can write other registers and it works as intended.  As we can download only the latest SD card image from the github, we do not know the version of history of the image, which one works and which one doesn't.

Thanks for your help.
-Y

dNechita, 2020-02-07:
I'm looking into this now.

dNechita, 2020-02-07:
So it crashes. Doesn't hangs, right?
Could you please provide whatever the application outputs until the crash?
Do you get same results when using the installer of the latest release?
(installer: aditof-demo_setup_x64.exe from here: https://github.com/analogdevicesinc/aditof_sdk/releases/tag/v1.2.0)

yhanalog, 2020-02-07:
The error message was something like "failed to get frames..", video freezes, and then USB camera device not recognized anymore.
I tried with the standard installer and it failed in the same way. Sometime it happens right after enabling/disabling small signal removal, sometimes it happens when wring a threshold value, usually within few attempts.

yhanalog, 2020-02-28:
It turned out the the image I used for testing was not the latest one. The version with the problem was "dragonboard410c-latest-960d341.img" dated 10/28/2019. 
The latest version I downloaded from the github as of today is "dragonboard-image-8de7aec.img" dated 1/16/2020. This code doesn't crash with simple register read and writes but it still crashes in certain conditions.
I have a code to write few registers and mode stop and mode start command are included as a part of the register writes. The software crashes with the following code with the error messages will be shown in the the next message. 
               uint16_t afeRegsAddr[REGS_CNT] = { 0x4001, 0x7c22, 0xc372, 0xc373, 0xc374, 0xc375, 0xc376, 0xc377, 0xc378, 0xc379, 0xc37a, 0xc37b, 0xc37c, 0xc37d, 0x4001, 0x7c22};
                uint16_t afeRegsVal[REGS_CNT] =  { 0x0006, 0x0004, 0x7888, 0xa997, 0x000a, 0x0157, 0x00c9, 0x007d, 0x0030, 0x0026, 0x006a, 0x0041, 0x0042, 0x001f, 0x0007, 0x0004};
                m_ctrl->writeAFEregister(afeRegsAddr, afeRegsVal, 16);

The software does not crash when I removed the mode commands as follows. Even though it doesn't crash, something doesn't seem quite right because all registers don't seem to be written correctly.
       m_ctrl->writeAFEregister(&afeRegsAddr[2], &afeRegsVal[2], 12);

The following is my test environment
- Rev B hardware
- Latest SD image for dragon boards as of day
- Visual Studio 2017, 
- Software is based on aditof-demo, Debug mode

yhanalog, 2020-02-28:
Visual Studio produces the following error message

Exception Thrown
Run-Time Check Failure #2 - Stack around the variable 'buf' was corrupted.

XinXuAnalog, 2020-03-02:
Maybe the reason why the 'buf' was corrupted is that we are writing 16 registers at the same time, which means the data size (64 Bytes) is larger than the buffer size (60 Bytes). When I tried writing 8 registers twice, I didn't really see this problem. Please correct me if I'm wrong. If this is the reason, I think you can just close this thread. Thanks.

dNechita, 2020-03-03:
I've looked into the issue and found the problem. The fix is on its way to master.
Indeed, the buffer size is 60 bytes but that shouldn't have been a problem since the data is split into smaller chunks and each chunk sent over USB.
The splitting action had a bug.

Thanks for reporting this!

closed 2020-03-09 by dNechita