analogdevicesinc/aditof_sdk #207: Demo: Missing #include <string> in system_impl.h of aditof C++ project

mockingjay14, 2020-02-11:
Problem: Unable to do building of aditof demo project in C++ despite doing default installation. Arising errors are shown in image below. 
[image]

Solution: Place **#include <string>** in the **system_impl.h** file located in the external dependencies of aditof project. 
[image]

dNechita, 2020-02-11:
You are right, thanks for reporting this.
We will add a fix.

dNechita, 2020-02-11:
Just out of curiosity, what compiler (and which version) did you use?
Thanks!

mockingjay14, 2020-02-12:
@dNechita I used Visual Studio 16 2019, Visual C++ v14.24.

closed 2020-02-12 by mockingjay14