analogdevicesinc/aditof_sdk #215: Create wiki page for Calibration Guide

grigorovici-andreea, 2020-02-20:


grigorovici-andreea, 2020-02-21:
https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/calibration

dstockhouse, 2020-03-04:
Hi, I'm using the ADI ToF eval kit for a university capstone project. The link to the calibration wiki above leads me to an error page. Has the page moved or been made non-public?
Thanks,
David

dNechita, 2020-03-05:
Hi David,

The page on the wiki website is currently under review and will be made public soon.
It's mostly based on the old doc: https://github.com/analogdevicesinc/aditof_sdk/blob/master/tools/calibration-96tof1/Calibration_Guide%20v2.pdf

dNechita, 2020-03-06:
Hi @dstockhouse 

The wiki page is now available at:
https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/calibration

closed 2020-03-09 by grigorovici-andreea