analogdevicesinc/aditof_sdk #225: Windows SDK Setup

davidvs-code, 2020-02-28:
I'm having issues with building the visual studio solution in the aditof_sdk windows user guide. Here is the error I get

C:\Users\dsala\aditof_sdk>cmake -DCMAKE_PREFIX_PATH="C:\projects\aditof-sdk\deps\glog\build_0_3_5\local_path\glog;C:\projects\aditof-sdk\deps\protobuf\build_3_9_0\local_path\protobuf;C:\projects\aditof-sdk\deps\libwebsockets\build_3_1\local_path\websockets" -G "Visual Studio 14 2015 Win64" -DOPENSSL_INCLUDE_DIRS="C:\OpenSSL-Win64\include" -DWITH_EXAMPLES=on ..
CMake Error: The source directory "C:/Users/dsala" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

Any ideas on what I could be doing wrong?

dNechita, 2020-02-28:
You need to create a build folder and then run the exact same command from there.

davidvs-code, 2020-02-28:
I tried that and got this error,

C:\Users\dsala\aditof_sdk\build>cmake -DCMAKE_PREFIX_PATH="C:\projects\aditof-sdk\deps\glog\build_0_3_5\local_path\glog;C:\projects\aditof-sdk\deps\protobuf\build_3_9_0\local_path\protobuf;C:\projects\aditof-sdk\deps\libwebsockets\build_3_1\local_path\websockets" -G "Visual Studio 14 2015 Win64" -DOPENSSL_INCLUDE_DIRS="C:\OpenSSL-Win64\include" -DWITH_EXAMPLES=on ..
-- Selecting Windows SDK version  to target Windows 10.0.18363.
CMake Error at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find Protobuf (missing: Protobuf_LIBRARIES Protobuf_INCLUDE_DIR)
  (Required is at least version "3.9.0")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindProtobuf.cmake:626 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  sdk/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/dsala/aditof_sdk/build/CMakeFiles/CMakeOutput.log".

I have a related question. The way I ran the setup steps was to create new directories for each step so that I have "glog" , "libwebsockets" , "protobuf" , and "aditof_sdk" folders in my user folder "dsala". Is that the correct process?

grigorovici-andreea, 2020-03-02:
Hello,

You are using the incorrect path, replace "C:\projects\", with your path, which is "C:\Users\dsala".

Also, for the ease of use, please refer to **https://github.com/analogdevicesinc/aditof_sdk/tree/master/scripts/windows**.
The script from this location builds the sdk and does all the necessary steps (including the download and installation of the dependencies).

davidvs-code, 2020-03-02:
Thanks for the info. A few questions,

1. If I instead try to get the SDK working by using the script, do I have to have Visual Studio 15 2017 Win 64 running on my PC? I currently have Microsoft Visual Studio Professional 2015 Version 14.0.25431.01 Update 3
2. How do I run the script? Do I copy and paste it into a new Visual Studio project and run it?

grigorovici-andreea, 2020-03-03:
Hello, 

To run the script you can just double click on it (in this way, the script will run with the default parameters, described in the documentation), or run it from a Command Prompt which is available in most Windows operating systems (in this way you can change the parameters).

Also, as a generator, you can use either "Visual Studio 15 2017 Win64" or "Visual Studio 14 2015 Win64".  With your version of Microsoft Visual Studio, I think the generator "Visual Studio 14 2015 Win64" should work.

I will let here an example of running the script in Command Prompt with generator **Visual Studio 14 2015 Win64**:

**setup_project.bat -g "Visual Studio 14 2015 Win64"**

Also, for more information about the parameters, please use -h:

**setup_project.bat -h**

davidvs-code, 2020-03-04:
Here is what I attempted,

C:\Users\dsala\aditof_sdk\scripts\windows>setup_project.bat -g "Visual Studio 14 2015 Win64"
Setup will continue with the configuration: Release
Setup will continue with the generator: "Visual Studio 14 2015 Win64"
The sdk will be built in: C:\Users\dsala\aditof_sdk\scripts\windows\build
The deps will be downloaded in: C:\Users\dsala\aditof_sdk\scripts\windows\deps
The deps will be installed in:  C:\Users\dsala\aditof_sdk\scripts\windows\deps\installed
Do you want to continue? [Y/N]y
Please enter OpenSSL root absolute path: C:\Users\dsala\libwebsockets\lib
C:\Users\dsala\libwebsockets\lib
Please enter opencv 3.4.1 root absolute path: C:\Users\dsala\opencv
C:\Users\dsala\opencv

Here is the output,

"Installing glog with config=Release and generator="Visual Studio 14 2015 Win64""
HEAD is now at a6a166d Update the version in CMakeLists.txt to 0.3.5
-- Selecting Windows SDK version 10.0.14393.0 to target Windows 10.0.18363.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/dsala/aditof_sdk/scripts/windows/deps/glog/build_0_3_5
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

  glog.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\glog\build_0_3_5\Release\glog.lib
  demangle_unittest.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\glog\build_0_3_5\Release\demangle_unittes
  t.exe
  logging_unittest.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\glog\build_0_3_5\Release\logging_unittest.
  exe
  stl_logging_unittest.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\glog\build_0_3_5\Release\stl_logging_u
  nittest.exe
  utilities_unittest.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\glog\build_0_3_5\Release\utilities_unitt
  est.exe
  -- Install configuration: "Release"
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/lib/glog.lib
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/include/glog/config.h
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/include/glog/logging.h
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/include/glog/raw_logging.h
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/include/glog/stl_logging.h
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/include/glog/vlog_is_on.h
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/include/glog/log_severity.h
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/lib/cmake/glog/glog-config.cmake
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/lib/cmake/glog/glog-config-version.cmake
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/lib/cmake/glog/glog-targets.cmake
  -- Installing: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/glog/lib/cmake/glog/glog-targets-release.cmak
  e
"Installing protobuf with config=Release and generator="Visual Studio 14 2015 Win64""
-- Selecting Windows SDK version 10.0.14393.0 to target Windows 10.0.18363.
--
-- 3.9.0.0
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/dsala/aditof_sdk/scripts/windows/deps/protobuf/build_3_9_0
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

  libprotobuf.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\protobuf\build_3_9_0\Release\libprotobuf.lib
  libprotobuf-lite.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\protobuf\build_3_9_0\Release\libprotobuf-l
  ite.lib
  libprotoc.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\protobuf\build_3_9_0\Release\libprotoc.lib
  protoc.vcxproj -> C:\Users\dsala\aditof_sdk\scripts\windows\deps\protobuf\build_3_9_0\Release\protoc.exe
  -- Install configuration: "Release"
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/protobuf/lib/libprotobuf-lite.lib
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/protobuf/lib/libprotobuf.lib
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/protobuf/lib/libprotoc.lib
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/protobuf/bin/protoc.exe
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/protobuf/lib/pkgconfig/protobuf.pc
  -- Up-to-date: C:/Users/dsala/aditof_sdk/scripts/windows/deps/installed/protobuf/lib/pkgconfig/protobuf-lite.pc
[... 451 lines cut ...]
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Error at examples/aditof-demo/CMakeLists.txt:13 (message):
  find_package(openCV) failed.  OpenCV_INCLUDE_DIR and OpenCV_LIBS need to be
  set.

-- Configuring incomplete, errors occurred!
See also "C:/Users/dsala/aditof_sdk/scripts/windows/build/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: ALL_BUILD.vcxproj
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: copy-dll-opencv.vcxproj
Microsoft (R) Build Engine version 14.0.25420.1
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: copy-dll-example.vcxproj

grigorovici-andreea, 2020-03-04:
Hello,

You need first to have OpenCV (best to have version 3.4.1) and OpenSSL installed, and when you are asked to give the absolute path to them the path should give the path to them. (In **readme.md** you can find the steps to be done before running the script: https://github.com/analogdevicesinc/aditof_sdk/tree/master/scripts/windows) 

From what I can see you put the path to libwebsockets instead of openSSL:
_"Please enter OpenSSL root absolute path: C:\Users\dsala\libwebsockets\lib
C:\Users\dsala\libwebsockets\lib
Please enter opencv 3.4.1 root absolute path: C:\Users\dsala\opencv
C:\Users\dsala\opencv"_

(The path to OpenSSL should look something like: C:\Users\dsala\ **OpenSSL-Win64** )

Also, make sure you have the OpenCV environment variables set, as stated in **readme.md**: https://github.com/analogdevicesinc/aditof_sdk/tree/master/scripts/windows.

grigorovici-andreea, 2020-03-06:
Hello,

Your visual studio solution explorer should look something like: 
[image]

Try to set up as a start-up project aditof-demo as a first example. (like in the screenshot above).

Also, when you run the project, make sure it is in the right configuration (Release or Debug).
In vs the default configuration is Debug, and if you did the same configurations for the script, as is the previous message you sent, then you have the Release configuration.

[image]

Also, please check in the project property page in vs if the Additional Include Directories are sent:
[image]

Thank you

davidvs-code, 2020-03-06:
I tried what you suggested and it's still not working. Here is my screenshot,

https://user-images.githubusercontent.com/58014982/76066069-b0168400-5f41-11ea-964e-8d9f79407ed3.png

grigorovici-andreea, 2020-03-09:
Hello,

You did not set-up your start-up project. **aditof-demo** should be **bolded**, not All build.

See in this screenshot how to do it:

[image]

grigorovici-andreea, 2020-03-09:
Hello,

Also if you want to build all. Then make sure that when the **ALL Build** is set as Start-up project  then the Configuration is set to release:

[image]

davidvs-code, 2020-03-10:
I tried doing what you suggested and it's still not working

https://user-images.githubusercontent.com/58014982/76282118-03dfe080-6254-11ea-965c-248bbc64f9b0.png

How do I know whether the configuration is debug or release? I tried both and neither work.

grigorovici-andreea, 2020-03-11:
Hello,

It is release configuration because when you ran the script you chose configuration **Release**. 

Can you please send me a screenshot with the directory structure should look like:
[image]

And inside the build folder:
[image]

And inside the deps/installed/libwebsockets folder:
[image]

Also, when you ran the script did you received any error message?

With this configuration, everything should work. So to find the problem we need to investigate more deeply, this is why I need all this information.

davidvs-code, 2020-03-11:
I deleted the "build" and "deps" folders. Then, because I think part of the problem is the libwebsockets, I ran the following lines in the command prompt again.

git clone --branch v3.1-stable --depth 1 https://github.com/warmcat/libwebsockets
cd libwebsockets
mkdir build_3_1 && cd build_3_1
cmake -DOPENSSL_ROOT_DIR="C:\OpenSSL-Win64" -DCMAKE_INSTALL_PREFIX=./local_path/websockets -G "Visual Studio 14 2015 Win64" ..
cmake --build . --target install --config Debug
cmake --build . --target install --config Release

The "libwebsockets" is now in the same folder that I have "aditof_sdk" in.

Then I ran this line of code in the "aditof_sdk/scripts/windows" folder,

setup_project.bat -g "Visual Studio 14 2015 Win64"

Then, with the hardware connected, in VS, I right clicked on "aditof-demo" in the solution explorer and selected, "Set as StartUp Project". Then I selected Release in the top toolbar of VS, x64, and clicked on the green arrow next to Local Windows Debugger and I got these errors

https://user-images.githubusercontent.com/58014982/76466525-1ddd0880-63a5-11ea-98fa-47a8d9f79c66.png

Here is the screenshot of the windows folder,

[image]

Here is a screenshot of the build folder,

https://user-images.githubusercontent.com/58014982/76466746-7b715500-63a5-11ea-9a3d-fd43d65e9107.png

Here is inside of libwebsockets folder

https://user-images.githubusercontent.com/58014982/76467332-a019fc80-63a6-11ea-8ec5-fbcafb9fb130.png

Here is the screenshot of the OpenSSL folder,

https://user-images.githubusercontent.com/58014982/76452706-c764cf80-638e-11ea-8d22-0ceb404ae5be.png

grigorovici-andreea, 2020-03-17:
Hello,

The first part you mentioned is unnecessary ("_Then, because I think part of the problem is the libwebsockets, I ran the following lines in the command prompt again._"). The script does that step and installs the libwebsockets into the **deps/installed** folder.

**When you run the script, do you get any error message?**  Please give a screenshot with the final messages from the command line where you run the script. 

**Do you have Windows 10?** 
**Is your system 64-bit?** If so, did you installed OpenSSL for Win64? 
[image]

For OpenCV did you installed at least version 3.4.1. And you set the environment variables?
[image]

Also from the Visual Studio solution, please first press the **rebuild** button and tell me if you get any errors. (do not press the green arrow first, let's see if the build passes, then if you can run the project.)
[image]

Thank you

davidvs-code, 2020-03-18:
Here is the error message I got when I tried Build > Rebuild Solution,

1>------ Rebuild All started: Project: ZERO_CHECK, Configuration: Release x64 ------
1>  Checking Build System
2>------ Rebuild All started: Project: aditof, Configuration: Release x64 ------
2>  Running cpp protocol buffer compiler on src/buffer.proto
2>  Building Custom Rule C:/Users/dsala/aditof_sdk/sdk/CMakeLists.txt
2>  calibration.cpp
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(20): warning C4267: '=': conversion from 'size_t' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(23): warning C4267: '=': conversion from 'size_t' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(26): warning C4267: '=': conversion from 'size_t' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(29): warning C4267: '=': conversion from 'size_t' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(31): warning C4244: '=': conversion from 'float' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(33): warning C4244: '=': conversion from 'float' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(191): warning C4244: 'argument': conversion from 'const uint32_t' to 'const float', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(195): warning C4244: 'argument': conversion from 'const uint32_t' to 'const float', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(206): warning C4244: 'argument': conversion from 'float' to 'std::size_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(237): warning C4244: 'argument': conversion from 'float' to 'std::size_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(252): warning C4244: '=': conversion from 'float' to 'uint32_t', possible loss of data
2>C:\Users\dsala\aditof_sdk\sdk\src\calibration.cpp(265): warning C4244: '=': conversion from 'float' to 'uint32_t', possible loss of data
2>  camera_96tof1.cpp
2>C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): warning C4251: 'aditof::Frame::m_impl': class 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::Frame'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): note: see declaration of 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>C:\Users\dsala\aditof_sdk\sdk\src\camera_96tof1.cpp(71): warning C4305: '=': truncation from 'double' to 'float'
2>C:\Users\dsala\aditof_sdk\sdk\src\camera_96tof1.cpp(72): warning C4305: '=': truncation from 'double' to 'float'
2>  camera_96tof1_specifics.cpp
2>  camera_chicony.cpp
2>C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): warning C4251: 'aditof::Frame::m_impl': class 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::Frame'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): note: see declaration of 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  camera_chicony_specifics.cpp
2>  camera_factory.cpp
2>  device_enumerator_ethernet.cpp
2>C:\Users\dsala\aditof_sdk\scripts\windows\deps\installed\protobuf\include\google/protobuf/map_type_handler.h(493): warning C4800: 'google::protobuf::uint64': forcing value to bool 'true' or 'false' (performance warning)
2>c:\users\dsala\aditof_sdk\sdk\src\network.h(17): fatal error C1083: Cannot open include file: 'libwebsockets.h': No such file or directory
2>  device_enumerator_factory.cpp
2>  device_enumerator_impl.cpp
2>  device_factory.cpp
2>  ethernet_device.cpp
2>C:\Users\dsala\aditof_sdk\scripts\windows\deps\installed\protobuf\include\google/protobuf/map_type_handler.h(493): warning C4800: 'google::protobuf::uint64': forcing value to bool 'true' or 'false' (performance warning)
2>c:\users\dsala\aditof_sdk\sdk\src\network.h(17): fatal error C1083: Cannot open include file: 'libwebsockets.h': No such file or directory
2>  filters_factory.cpp
2>  frame.cpp
2>C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): warning C4251: 'aditof::Frame::m_impl': class 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::Frame'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): note: see declaration of 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  frame_impl.cpp
2>  frame_operations.cpp
[... 252 lines cut ...]
3>  Generating Code...
8>LINK : fatal error LNK1181: cannot open input file '..\..\sdk\Release\aditof.lib'
6>LINK : fatal error LNK1181: cannot open input file '..\..\sdk\Release\aditof.lib'
3>LINK : fatal error LNK1181: cannot open input file '..\..\sdk\Release\aditof.lib'
10>------ Rebuild All started: Project: ALL_BUILD, Configuration: Release x64 ------
[... cut ...]
https://user-images.githubusercontent.com/58014982/76919715-e976ba00-6886-11ea-9a0c-01e00698de0b.png

Also, to answer your questions,

1. Yes, I do have Windows 10 and my system is 64-bit
2. Yes, the OpenSSL for Win64 version I have installed is the one you indicated in your screenshot
3. Here is a screenshot of my environment variables

https://user-images.githubusercontent.com/58014982/76919996-a406bc80-6887-11ea-90ee-cf7699dca18a.png

Thanks for your persistence in helping me to get this to work.

davidvs-code, 2020-03-18:
Also, I forgot to mention that I have OpenCV version 3.4.8

davidvs-code, 2020-03-18:
Also, I tried to reinstall OpenSSL in case I did something wrong the first time. I'm choosing "The windows system directory" in the screenshot. Is that correct?

https://user-images.githubusercontent.com/58014982/76920840-b8e44f80-6889-11ea-859e-d658d2f77ddb.png

dNechita, 2020-03-18:
Try the other option (The OpenSSL binaries (/bin) directory.

grigorovici-andreea, 2020-03-18:
Hello,

Looks like the dependencies are correct, do you have the possibility to do the steps on a clean set-up?

Also, can you show me the output of the script? (Delete **deps** and **build** folder, run the script again, and show me the last 25 lines that the script outputs).

Thanks

davidvs-code, 2020-03-18:
I reinstalled OpenSSL with the binaries (/bin) directory option. I deleted deps and build then tried building aditof_demo again and got the following error (only showing the last few lines)

>C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): warning C4251: 'aditof::Frame::m_impl': class 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::Frame'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): note: see declaration of 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>C:\Users\dsala\aditof_sdk\sdk\include\aditof/system.h(73): warning C4251: 'aditof::System::m_impl': class 'std::unique_ptr<SystemImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::System'
2>          with
2>          [
2>              _Ty=SystemImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/system.h(73): note: see declaration of 'std::unique_ptr<SystemImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=SystemImpl
2>          ]
2>  main.cpp
2>C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): warning C4251: 'aditof::Frame::m_impl': class 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::Frame'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/frame.h(73): note: see declaration of 'std::unique_ptr<FrameImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=FrameImpl
2>          ]
2>C:\Users\dsala\aditof_sdk\sdk\include\aditof/system.h(73): warning C4251: 'aditof::System::m_impl': class 'std::unique_ptr<SystemImpl,std::default_delete<_Ty>>' needs to have dll-interface to be used by clients of class 'aditof::System'
2>          with
2>          [
2>              _Ty=SystemImpl
2>          ]
2>  C:\Users\dsala\aditof_sdk\sdk\include\aditof/system.h(73): note: see declaration of 'std::unique_ptr<SystemImpl,std::default_delete<_Ty>>'
2>          with
2>          [
2>              _Ty=SystemImpl
2>          ]
2>  Generating Code...
2>LINK : fatal error LNK1181: cannot open input file '..\..\sdk\Release\aditof.lib'
========== Build: 0 succeeded, 2 failed, 1 up-to-date, 0 skipped ==========

I'm also trying to get a different machine to try this on.

grigorovici-andreea, 2020-03-20:
Hello,

Please check the tutorial that we created, and follow the steps presented:  https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/windows/build_instructions.md

Thank you

grigorovici-andreea, 2020-03-24:
Hello @davidvs-code,

Did you manage to compile the project?

Thank you.

davidvs-code, 2020-04-05:
I got the solution to build using a brand new machine (Windows 10, 64 bit). However I'm now getting this error when trying to run aditof-demo in Visual Studio Community 2015,

https://user-images.githubusercontent.com/58014982/78464565-d42fc700-769f-11ea-8227-342c23355446.png

Here is a screenshot of how I've setup my environment variables.

https://user-images.githubusercontent.com/58014982/78464631-6a63ed00-76a0-11ea-92e1-1c2ce9db0702.png

I don't understand why it's not working. opencv_world348.dll is in the opencv\build\x64\vc14\bin directory

grigorovici-andreea, 2020-04-07:
Hello,

Can you please check if opencv_world348.lib is set in the Properties windows -> Linker -> Input -> Additional dependencies. (Here is an example how to get there:)

[image]
[image]

Also as a quick fix solution just copy that dll that is missing (opencv_world348.dll) into: **your_path\aditof_sdk\scripts\windows\build\examples\aditof-demo\Release**.

(Something like that: )

[image]

Thank you

davidvs-code, 2020-04-08:
Got it working! Thanks a lot for your help and for your patience. Any tips on the best way to record video. I didn't see it as an example.

grigorovici-andreea, 2020-04-08:
Hello,

You can check this out: https://github.com/analogdevicesinc/aditof_sdk/tree/master/examples/aditof-demo#saving-data-to-a-file.

Or you can take a look into the recorder class from: https://github.com/analogdevicesinc/aditof_sdk/blob/master/examples/aditof-demo/aditofdemorecorder.cpp. In the demo example, we can actually also record videos (save the frames to load later).  
[image]

grigorovici-andreea, 2020-04-15:
Hello @davidvs-code,

As this is working fine for you now, can we close the issue?

Thank you

davidvs-code, 2020-04-15:
Yes of course, thanks a lot for your help.

closed 2020-04-15 by davidvs-code