analogdevicesinc/aditof_sdk #245: Open3D follow up point cloud enhancements

grigorovici-andreea, 2020-03-09:
Follow up for issue: 
https://github.com/analogdevicesinc/aditof_sdk/issues/175

Still waiting for an answer from: intel-isl/Open3D#1516.

grigorovici-andreea, 2020-03-18:
The answer received from the Open3D team. The pyramidal form is correct. 

There is still a problem left: all the strait objects appear to be curved in the point cloud. Will check with the ROS results to see if is a camera problem, or a point cloud problem.

grigorovici-andreea, 2020-03-25:
> The answer received from the Open3D team. The pyramidal form is correct.
> 
> There is still a problem left: all the strait objects appear to be curved in the point cloud. Will check with the ROS results to see if is a camera problem, or a point cloud problem.

the mentioned problem (strait objects appear curved) will be solved with: #274

closed 2020-03-25 by grigorovici-andreea

jwrl7, 2021-03-31:
Hello, could someone elaborate on what the purpose of the pyramidal form is. 
Im working with the sensor and after calibrating both the intrinsic and distortion parameters, I still see strange distortion in the depth image as described "strait objects appear to be curved"

Thanks!!

acozma, 2021-04-14:
Are you using your own correction equation for the lens distortion or the [function in the SDK](https://github.com/analogdevicesinc/aditof_sdk/blob/master/sdk/src/cameras/ad-96tof1-ebz/calibration_96tof1.cpp#L439)?