analogdevicesinc/aditof_sdk #256: Video stream fails to open - SDK Debian 10 Linux

acvin, 2020-03-15:
Hi, 
Compiled sdk with examples from 
[ADI sdk Linux](https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/linux/user_guide.md)
for my Debian 10 x86_64 machine. Compilation went fine and camera is detected in dmesg as follows:

> usb 1-2: Product: Android
> usb 1-2: Manufacturer: Android
> usb 1-2: SerialNumber: 15691762
> usb 1-2: USB disconnect, device number 7
> usb 1-2: new high-speed USB device number 8 using xhci_hcd
> usb 1-2: New USB device found, idVendor=1d6b, idProduct=0102, bcdDevice= 4.09
> usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> usb 1-2: Product: ADI TOF DEPTH SENSOR
> usb 1-2: Manufacturer: Linux Foundation
> uvcvideo: Found UVC 1.00 device ADI TOF DEPTH SENSOR (1d6b:0102)

When I run the first-frame app (or the adi demo one), console shows the following error and image popup is not shown

> ./first-frame 
> I0315 10:31:23.242889   415 device_enumerator_linux.cpp:48] Looking for USB connected devices
> I0315 10:31:23.473628   415 system_impl.cpp:62] System initialized
> I0315 10:31:23.473758   415 camera_96tof1.cpp:78] Initializing camera
> I0315 10:31:23.473793   415 usb_device_linux.cpp:115] Opening device
> I0315 10:31:24.266456   415 calibration.cpp:261] EEPROM calibration data size 65044 bytes
> I0315 10:31:30.901198   415 camera_96tof1.cpp:95] Camera initialized
> I0315 10:31:30.903856   415 camera_96tof1.cpp:129] Chosen mode: near
> I0315 10:31:30.903911   415 camera_96tof1.cpp:170] Camera range for mode: near is: 250 mm and 800 mm
> I0315 10:31:30.904278   415 camera_96tof1.cpp:180] Found firmware for mode: near
> I0315 10:31:30.904289   415 camera_96tof1.cpp:183] Firmware size: 10440 bytes
> W0315 10:31:38.380816   415 usb_device_linux.cpp:357] VIDIOC_STREAMON, error:110(Connection timed out)
> W0315 10:31:38.381404   415 camera_96tof1.cpp:188] Failed to program AFE
> E0315 10:31:38.381484   415 main.cpp:88] Could not set camera mode!

Any ideas? 
Thanks 
Luca

dNechita, 2020-03-16:
I'm still trying to reproduce.
Does the timeout error happen again if you reboot the dragonboard401c?
Which SD card image are you using?

dNechita, 2020-03-17:
I tried this on a fresh installed Debian 10.3 and things seem to work fine.
Can you update the SD card with the latest image? See here: https://github.com/analogdevicesinc/aditof_sdk/blob/master/README.md

acvin, 2020-03-17:
Hi Dan, 
thank you for your support. 
I tried to reboot and restart the dragonboard401c several times, actually I managed to get it working few times. Not sure what is the issue, but it does not seem to be related to the camera itself. 
How can I check if I have the latest SD image revision? 
Thanks

dNechita, 2020-03-18:
Hi Luca,
You can only know the SD card image revision if you still have the adapter of the SD card that was shipped with the box. It's written on it (e.g. ADITOF-SDK v1.1.1). Or if you downloaded a SD card image at some point in time, you can check the name of the image after it was extracted from its archive.
The USB connection can be unstable sometimes. Maybe try with another usb cable or add a usb hub between the PC and the board.

closed 2020-03-18 by acvin