analogdevicesinc/aditof_sdk #262: Investigate maskr-cnn

dNechita, 2020-03-17:


grigorovici-andreea, 2020-04-01:
Some intermediate results. 

[image]

grigorovici-andreea, 2020-04-08:
Investigation results:

- we need to give access to 3 configuration files: **COCO object class LABELS**, **TENSOFLOW model configuration files, based on COCO dataset**: _frozen_inference_graph.pb and mask_rcnn_inception_v2_coco_2018_01_28_ 
We can either give access directly to the files, by adding them in our repo or give the user the following link from where they can use the script to get access to any model:(https://github.com/opencv/opencv_extra/blob/master/testdata/dnn/download_models.py).

- the results are reasonable good:
[image]

- for real-time processing and results using mask-RCNN is not feasible as the time for processing a frame is around 1.50 seconds:
[image]
 * the time for processing a frame si not influenced by our data, any picture I gave to the script the time did not improve. 
 * also checked by getting a video from youtube to be processed, the same result:
[image]

closed 2020-04-23 by grigorovici-andreea