analogdevicesinc/aditof_sdk #284: Problem with "USB bridge" under Linux (Ubuntu in a virtual machine)

arrowperfecto, 2020-03-31:
Using image released to support resolving issue [#217](https://github.com/analogdevicesinc/aditof_sdk/issues/217).  I’ve also tried to use the “usb bridge” under linux (Ubuntu in a virtual machine): 
The camera is recognized by the pc:
[image]
I tried the script “first_frame.py”, the initialization seems ok but it failed to get a frame…  
[image]
Are you aware of this kind of issue under linux (and/or with a virtual machine)?
Thank you.

dNechita, 2020-03-31:
Hi,
On what host is the Virtualbox running? Is it Windows 10?

arrowperfecto, 2020-03-31:
Yes, it’s windows 10

dNechita, 2020-03-31:
I didn't encountered this error before. Things should work on Linux (non VM).
Also things should work on linux virtual machines (mostly Ubuntu and Debian) but the difference is that the host was an Ubuntu 18.04 not Windows. (using Virtual Machine Manager)

dNechita, 2020-04-01:
I did some tests using VirtualBox on Windows with a Ubuntu 18.04 VM. The USB communication is pretty unstable and examples don't work.
I don't recommend using this type of setup.

closed 2020-05-27 by acozma