analogdevicesinc/aditof_sdk #299: Missing Device Tree for IMX8M

bitslip6, 2020-04-10:
Hi!

First I would like to thank you for all of the great work in the last few months in this repo.  The updates have been consistent and always improving and it shows.  So great work!

I am close to having the dev kit working against an NXP IMX8M Mini board.  I have a yocto build system (I can send you bit bake recipes if you like) and the addi9036 driver built as a kernel module against 4.14.78 using the source provided [here](https://github.com/analogdevicesinc/linux/commit/e6942ce7c41d16d4070a3825f1b1e0491043fb8b).  The EPROM is connected to i2c bus 3 and shows devices 56 and 57 with "i2cdetect 3".  The mipi cable is connected via a break out board and gnd to i2c bus 1.  we are using 1.8v and have JP1 soldered to pins 1&2.  All other solder pins are connected as for raspberry pi.

I see device 0x64 on i2c bus 1.

after loading the kernel module and setting the addi9036 on i2c-1 with this command:
`echo addi9036 100 > /sys/bus/i2c/devices/i2c-1/new_device
`

I receive the following kernel messages:
```
[  165.909817] addi9036 1-0064: addi9036_probe: i2c addr = 0x64
[  165.914310] addi9036 1-0064: endpoint node not found
[  165.918094] addi9036: probe of 1-0064 failed with error -22

```

some investigation reveals that the open firmware pointer on the *device is NULL and I believe this line: https://elixir.bootlin.com/linux/latest/source/include/linux/device.h#L794
```
static inline struct device_node *dev_of_node(struct device *dev) {
	if (!IS_ENABLED(CONFIG_OF) || !dev)
		return NULL;
	return dev->of_node;
}
```

is returning null because I am missing a valid device tree for the camera.  If you could provide the device tree for raspberry pi, I would be happy to port to the IMX8 Mini and send pull request.

Any help you can provide would be very appreciated!!
Best Regards,
Cory

btogorean, 2020-04-13:
Hi Croy,

Thank you for your appreciation and for your contribution. 
It really means a lot to us as we try to extend our supported boards list and your work on I.MX8 porting is helping us a lot.

We are sorry that our Linux kernel repository is not so intuitive and branch splitted but the main reason is that we rebase our work on top of Xilinx Linux repo and Raspberry Pi.

So the devicetree that we use on RPi is this one: [rpi-addi9036-overlay.dts](https://github.com/analogdevicesinc/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/rpi-addi9036-overlay.dts) on _rpi-4.19.y_ branch

I also updated the documentation page with a link to the devicetree.

If we can help you with other information please don't hesitate to contact us via this channel or using Engineer Zone Forum.

Regards,
Bogdan

bitslip6, 2020-04-14:
Thanks so much for the speedy reply!

I think this is the last step to having the board working on the NXP imx8 mini.  I'll check with or project lead but I don't think there would be any issue providing yocto recipes to get everything running on other custom boards as well.

I'll let you know when I have some more progress. and hopefully i can send over a pull request soon.

Best regards,
Cory

bitslip6, 2020-04-24:
After some further investigation on the EEPROM, it looks like 0x56 is page 0 of the atmel 24c1024 and 0x57 is page 1 of the atmel 24c1024 however the device tree you linked only maps 0x56 to the 24c kernel driver.

question.  should the kernel handle both 0x56 and 0x57 or are you reading/writing page 1 on 0x57 from userspace?   If you are using the kernel driver, should I add device tree info for 0x57?

I hope this is clear.   Basically I just want to know if the kernel 24c driver should be handling both 0x56 and 0x57.

Thanks much!!!!

btogorean, 2020-04-29:
Hi,

We are using the at24.c driver from linux kernel to access the EEPROM memory.
From what I understand this driver already have support for 16 bit address chips and is implemented [here](https://github.com/torvalds/linux/blob/master/drivers/misc/eeprom/at24.c#L244)
We have the AT24_FLAG_ADDR16 flag enabled because we specify as compatible `24c1024`.

I don't think there is required to explicitly specify a second device on devicetree to access page 1.

Regards

bitslip6, 2020-04-30:
Thanks for the reply!

so I have access to the eeprom through the sysfs for device 0x56 (which is listed in my DTB) but I am unable to access device at address 0x57 through sysfs (which is not in my DTB).

Would there be any issue with adding device 0x57 to the DTB, or should i just do this manually when I need access to that eeprom page through the /sys/bus/i2c/devices/i2c-1/new_device interface?

Thanks for your reply!
Cory

btogorean, 2020-05-04:
Hi Cory,

The EEPROM device used on ADI ToF is an AT24CM01. This device should contain 1Mb of data.
Can you through sysfs on device 0x56 access 131,072 bytes of data?

If yes I think there is no problem and you don't need the second 0x57 to handle the second page of the EEPROM. The address bit that makes the difference between 0x56 and 0x57 it is actually the 17th bit (counting from 1) of the address. Addressing on 17 bits we can address 131,072 words of 8 bit each, exactly the size of this EEPROM.

Regards,
Bogdan

closed 2020-07-16 by acozma