analogdevicesinc/aditof_sdk #300: Linux SDK Setup

s9323249, 2020-04-13:
I'm having issues with building in the aditof_sdk Linux  linux scripts . 
https://github.com/analogdevicesinc/aditof_sdk/tree/master/scripts/linux/setup.sh
Here is the error I get
---------------------------------------------------------------------------------------------------
-- Build files have been written to: /home/fw/111/aditof_sdk-master/scripts/linux/build
[  2%] Building CXX object sdk/CMakeFiles/aditof.dir/src/network.cpp.o
[  4%] Building CXX object sdk/CMakeFiles/aditof.dir/src/device_enumerator_ethernet.cpp.o
[  7%] Building CXX object sdk/CMakeFiles/aditof.dir/buffer.pb.cc.o
[  9%] Building CXX object sdk/CMakeFiles/aditof.dir/src/ethernet_device.cpp.o
In file included from /home/fw/111/aditof_sdk-master/sdk/src/network.h:32:0,
                 from /home/fw/111/aditof_sdk-master/sdk/src/network.cpp:32:
/home/fw/111/aditof_sdk-master/scripts/linux/build/sdk/buffer.pb.h:10:40: fatal error: google/protobuf/port_def.inc: No such file or directory
 #include <google/protobuf/port_def.inc>
                                        ^
compilation terminated.
In file included from /home/fw/111/aditof_sdk-master/scripts/linux/build/sdk/buffer.pb.cc:4:0:
/home/fw/111/aditof_sdk-master/scripts/linux/build/sdk/buffer.pb.h:10:40: fatal error: google/protobuf/port_def.inc: No such file or directory
 #include <google/protobuf/port_def.inc>
                                        ^
compilation terminated.
In file included from /home/fw/111/aditof_sdk-master/sdk/src/network.h:32:0,
                 from /home/fw/111/aditof_sdk-master/sdk/src/device_enumerator_ethernet.cpp:33:
/home/fw/111/aditof_sdk-master/scripts/linux/build/sdk/buffer.pb.h:10:40: fatal error: google/protobuf/port_def.inc: No such file or directory
 #include <google/protobuf/port_def.inc>
                                        ^
compilation terminated.
In file included from /home/fw/111/aditof_sdk-master/sdk/src/network.h:32:0,
                 from /home/fw/111/aditof_sdk-master/sdk/src/ethernet_device.cpp:34:
/home/fw/111/aditof_sdk-master/scripts/linux/build/sdk/buffer.pb.h:10:40: fatal error: google/protobuf/port_def.inc: No such file or directory
 #include <google/protobuf/port_def.inc>
                                        ^
compilation terminated.
sdk/CMakeFiles/aditof.dir/build.make:646: recipe for target 'sdk/CMakeFiles/aditof.dir/buffer.pb.cc.o' failed
make[2]: *** [sdk/CMakeFiles/aditof.dir/buffer.pb.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
sdk/CMakeFiles/aditof.dir/build.make:214: recipe for target 'sdk/CMakeFiles/aditof.dir/src/device_enumerator_ethernet.cpp.o' failed
make[2]: *** [sdk/CMakeFiles/aditof.dir/src/device_enumerator_ethernet.cpp.o] Error 1
sdk/CMakeFiles/aditof.dir/build.make:166: recipe for target 'sdk/CMakeFiles/aditof.dir/src/network.cpp.o' failed
make[2]: *** [sdk/CMakeFiles/aditof.dir/src/network.cpp.o] Error 1
sdk/CMakeFiles/aditof.dir/build.make:382: recipe for target 'sdk/CMakeFiles/aditof.dir/src/ethernet_device.cpp.o' failed
make[2]: *** [sdk/CMakeFiles/aditof.dir/src/ethernet_device.cpp.o] Error 1
CMakeFiles/Makefile2:117: recipe for target 'sdk/CMakeFiles/aditof.dir/all' failed
make[1]: *** [sdk/CMakeFiles/aditof.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Any ideas on what I could be doing wrong?

dNechita, 2020-04-13:
Which version of the protobuf was detected?
It should be written a couple of lines above this one:
"-- Build files have been written to: /home/fw/111/aditof_sdk-master/scripts/linux/build"

s9323249, 2020-04-13:
i check  the protobuf version:
fw@fw-desktop:~$ protoc --version
libprotoc 3.9.0

dNechita, 2020-04-13:
But what version of protobuf does CMake detect?

acozma, 2020-05-27:
See issue #303

closed 2020-05-27 by acozma