analogdevicesinc/aditof_sdk #302: Rpi-3 SDK setup

littleyueh, 2020-04-13:
Hi,

I have tried to setup the SDK on my raspberry pi 3 model b+, with my Chicony camera module
so I take the following link as my reference to setup:
https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/raspberrypi3/chicony_user_guide.md

and I have few questions:
1.  Is it enough for me to only install the following packages, if I just want to get the raw data(IR image & DoF image) from camera(through MIPI to Rpi) and send it to my PC via Wi-fi?
[image]

2.  In the document I saw this:
[image]
May I ask where can I download the image file and put it on my Rpi-3?
Is the image file generated from building the Linux kernel?

3. I have tried to build kernel for Rpi-3, as the following steps
[image]
but at the end the kernel failed to start the module
I am wondering two points:
1) `export CROSS_COMPILE=< location of ARM toolchain >arm-linux-gnueabihf-`
should I install the toolchain from raspberry and add it to the path?
(https://wiki.analog.com/resources/tools-software/linux-build/generic/raspberrypi)
2) `cp ./arch/arm/zImage <sd card boot partition>/kernel7.img` 
the path in my file is ./arch/arm/boot/zImage, for your information

I will appreciate any help and advice, thanks

dNechita, 2020-04-13:
> 1. Is it enough for me to only install the following packages, if I just want to get the raw data(IR image & DoF image) from camera(through MIPI to Rpi) and send it to my PC via Wi-fi?
>    [image]

Yes, these are enough and they should be already available on the SD card image. Also, libopencv-dev is not required for what you are tying to do.

> 2. In the document I saw this:
>    [image]
>    May I ask where can I download the image file and put it on my Rpi-3?

You can find it in the readme file of the project: https://github.com/analogdevicesinc/aditof_sdk/blob/master/README.md. There's a table called 96-TOF1-EBZ which has a link in the Downloads column that oorresponds to Raspberry Pi.
>    Is the image file generated from building the Linux kernel?

Yes.
> 3. I have tried to build kernel for Rpi-3, as the following steps

@btogorean can you help?

littleyueh, 2020-04-13:
Hi,

According to your reply
Is it correct that I don't need to download those packages and build the Linux kernel, if I just simply download the SD card image that you mentioned

And after write the .img file to SD card, I can try the example code to fetch the image data.
I plan to use Python to get the image.
(https://github.com/analogdevicesinc/aditof_sdk/tree/master/bindings/python/examples)

Thanks for your reply
Best regards

dNechita, 2020-04-14:
> Is it correct that I don't need to download those packages and build the Linux kernel, if I just simply download the SD card image that you mentioned

Yes, that is correct.

btogorean, 2020-04-15:
> I am wondering two points:
>
> 1. `export CROSS_COMPILE=< location of ARM toolchain >arm-linux-gnueabihf-`
>    should I install the toolchain from raspberry and add it to the path? [...]

Hi,

related to point 1. the example is for cross compilation on other Host like your PC and then just copy the kernel and modules to RPi SD card. So < location of ARM toolchain > depends on your host configuration and if you installed linaro arm compiler I think you don't need the explicit path.

For point 2. you're right, the zImage will be in /boot subfolder. I'll create a PR to fix the documentation

littleyueh, 2020-04-17:
Hi, 

thanks for helping, I successfully setup the SDK

Unfortunately, it returns the following results:
[image]

I have changed two different chicony ToF modules, and also reset the interposer; however it returns same result

best regards

closed 2020-05-27 by acozma