analogdevicesinc/aditof_sdk #303: execute the Setup.sh scripts to proceed Linux setup fails

grigorovici-andreea, 2020-04-13:
Question attached.

[ROS Wrapper for the ADI ToF library question.xlsx](https://github.com/analogdevicesinc/aditof_sdk/files/4469726/ROS.Wrapper.for.the.ADI.ToF.library.question.xlsx)

grigorovici-andreea, 2020-04-13:
The problem might be that the script is downloading opencv by itself and the one that is already is system is interfering, please use the option --buildopencv  when running the script.  The script will download, build and install version 3.4.1 of opencv if the argument “—buildopencv” is given. Also before re-running the script, please delete the files already created by the script.

s9323249, 2020-04-14:
1.Delete the files already created by the script.
2.Re-running the script
**./setup.sh -bo**

get the error,
[image]

grigorovici-andreea, 2020-04-15:
Hello,

Can you please tell what Kernel do you have on your ubuntu? 

Also, when you first run the script can you tell us the output you receive, we need to know which protobuf is found by CMAKE. 

Also, do you know if you installed before protobuf using apt-get?

Thank you

s9323249, 2020-04-16:
Can you please tell what Kernel do you have on your ubuntu?
**>>4.15.0-96-generic** 

when you first run the script can you tell us the output you receive, we need to know which protobuf is found by CMAKE.
**>> please see attachment files. (log,txt)**

 do you know if you installed before protobuf using apt-get? 
**>>Only using script install.**

[log.txt](https://github.com/analogdevicesinc/aditof_sdk/files/4484488/log.txt)

grigorovici-andreea, 2020-04-22:
Hello,

We tried on a virtual machine and the script should work. 
In order to figure out the problem, can you please try to build the project using the building steps from here and let us know if any problem appears and where: **https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/linux/build_instructions.md.**

s9323249, 2020-04-22:
Dear,
I setp by step installation,and the screenshot,
**Glog:**
[image]
**Libwebsockets:**

![
    Libwebsockets](https://user-images.githubusercontent.com/63579578/79957719-96f77180-84b4-11ea-844d-2b6ca718a9eb.png)

**protobuf:**

[image]

**Download and build SDK only**
> make
an error appears,

[image]

grigorovici-andreea, 2020-04-24:
Hello,

As we cannot reproduce your problem, we ask you kindly to try again on a clean set up.
It looks like the dependencies are correct, but our only guess is that they interfere with something that was already installed on your system.
 Do you have the possibility to do the steps on a clean set-up? (that means on a new Linux where you do not have installed everything yet).

Also please make sure you clone the latest version from the aditof sdk repository.

s9323249, 2020-05-04:
Dear,
thanks for your advice, i try a new Linux ubuntu 18.04,
and it can be compiled.
and run the example  aditof- demo ,an error appears,
[image]

can you give some advice,thanks.

grigorovici-andreea, 2020-05-05:
Hello,

Do you have the camera connected through USB?
Can you check if your system recognize the camera? (use dmesg command in a new terminal, should appear under the name ADITOF DEPTH SENSOR).

Also, you can choose the connection you want with the camera (USB/Ethernet) by using the first checkbox.

Thank you?

s9323249, 2020-05-08:
hi,
thanks for your advice.
First i check the device can  correct work on windows 10 ,
[image]
[image]

and then run on Ubuntu system ,use dmesg command,
[image]

s9323249, 2020-05-08:
hi,
Do you have the camera connected through USB?
**>>Yes**
Can you check if your system recognize the camera? (use dmesg command in a new terminal, should appear under the name ADITOF DEPTH SENSOR).

grigorovici-andreea, 2020-05-11:
Hello,

It looks like the ADITOF DEPTH SENSOR is nor recognized by your system.

Please see our documentation from here: _https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/ug_linux_, the section **Enable the USB connection to the host PC**.

Thank you

grigorovici-andreea, 2020-05-18:
Hello @s9323249,

Did you succeed with the project? 

Is anything else we can help you with for this ticket? 

Thank you

s9323249, 2020-05-20:
Hello,
thanks for your advice.
Now having some problems,
Hardware :  Raspberry Pi 4 + chicony Tof
[image]

Through the interface USB3.0 connect   Raspberry Pi 4 + chicony Tof

And using  Lasest SD card image

[image]

Run the Aditof-Demo
[image]
[image]

But Can't find camera,can you give some advice for USB3.0 method,thanks.

dNechita, 2020-05-21:
Hi,

 As you can see in the log messages of the aditof-demo, it looks for the device (camera) on the target (MIPI connection).
To make it look for the devices connected on USB, you need to make sure you are building the entire project for a remote PC and not for a target. This is done when generating the build system with CMake.
Create a new build folder or delete the existing one then create an empty build folder. The run cmake but don't specify the target (don't add -DRASPBERRYPI=1 in the cmake command).

We haven't tried this on a target before, in theory it should work.

s9323249, 2020-05-22:
Hello,
thanks for your advice.
but we would like to know why we need to build entire project in PC rather than on the rasp, than currently we meet the error like the below: we run cmake command after delete build folder 

[image]

could you help us to solve this problem...thx.

dNechita, 2020-05-22:
There are 2 modes when you build the project: 'remote mode' and 'target mode' (The target mode is when you define a target on the cmake command, e.g. cmake -DRASPBERYPI=1. If you don't then the mode will be remote-mode). 
On the 'remote-mode' the project expects that camera are accessible through some channels of communication (USB and Ethernet).
On 'target-mode' the project expects that the camera is available on the system of the target (through MIPI connection).
You don't have to build the project (in remote-mode) only on PC, it can be done on RPI or other platforms that operate with a Linux distribution). It's just that the common scenario is to build it on a PC.

Regarding you build issue:
You need to tell CMake where to look for Glog, Protobuf and Websockets. It looks you have them at path:
/home/pi/workspace/github/aditof_sdk/deps/installed
So replace the paths referring to /opt.
E.g.
`DCMAKE_PREFIX_PATH="/home/pi/workspace/github/aditof_sdk/deps/installed/glog;/home/pi/workspace/github/aditof_sdk/deps/installed/protobuf;/home/pi/workspace/github/aditof_sdk/deps/installed/websockets"`

closed 2020-07-16 by acozma