analogdevicesinc/aditof_sdk #310: [Jetson Nano] imshow-jetson configuration

KamelAbdelouahab, 2020-04-21:
Hi all,

I'm testing the ADDI9036 board with a Jetson Nano connected through MIPI. I compiled the kernel and device tree according to [these instructions](https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/ug_jetson), and used the provided [script]{https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/jetson/build_instructions.md) to build the software suite.

After this, I found that the imshow-jetson example doesn't capture frames. It also returns zeros frames sometimes for a reason I can't figure out. Surprisingly, running the "first-frame" example seems to fix the problem: the imshow-jetson works fine after this ...

Is there a reason/quick fix for for this ? 

Best,
Kamel

btogorean, 2020-04-22:
Hi Kamel,

Did you tried the SD card image we provided? does it have the same strange behavior? [SD-Image](http://swdownloads.analog.com/cse/aditof/jetson-latest.tar.xz)

KamelAbdelouahab, 2020-04-22:
Thansk @btogorean ,

I gave the image a try, but it size exceeded the capacity of my 32GB SD card (the only one available for now, thanks Covid19 )
I tried resizing using qemu-img, but it seems that I corrupted the image. Booting on the resized image on the Nano only displays a basic bash with no oem-config on the device ... 

I hope I can get larger SD to test the image in the next days though.

dNechita, 2020-04-22:
Hi,
In case your are using dd to write the image to the SD card, try to limit the number of bytes to be written to the SD card.
E.g.
dd if=name-of-the-image.img of=/dev/xxx bs=4M count=7680
(In the example above, I'm writting exactly 30GB, which is 30720MB and since the block size is 4MB, then number of blocks to be written becomes 30720MB /4)

KamelAbdelouahab, 2020-04-23:
Thanks for your answers. We'll be testing this in the next days.

closed 2020-05-27 by acozma