analogdevicesinc/aditof_sdk #314: Not able to use a IMX219 camera with Jetson Nano

adry84, 2020-04-23:
Hi,

I want to use a IMX219 device on the slot CSI0 and an ADITOF camera on the slot CSI1 on Jetson Nano.
I see that the IMX219 support was removed from [the device tree.](https://github.com/analogdevicesinc/aditof_sdk/blob/master/misc/nvidia/jetson/kernel_src/hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-porg-p3448-common.dtsi)
Is there a reason? Is that possible to get it back?

Regards,

Audrey

btogorean, 2020-04-24:
Hi Audrey,

Disabling IMX219 actually was a side effect of disabling plugin manager. According to Nvidia [information](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fcamera_sensor_prog.html%23wwpID0E0QG0HA) plugin manager should be disabled if connected camera EEPROM does not store a valid camera ID.

Also we used model A02 carrier board for our tests that has only one CSI interface. Unfortunately at this moment we don't have a B01 carrier with two CSI ports to perform tests on.

I think the solution for you would be to add to the platform devicetree the IMX219 camera specifying its connection to CSI0 port. 
The main disadvantage is that this is a fixed configuration and not dynamical as plugin manager would allow it but given the current context I think is the only possibility.

Also unfortunately we can't create this devicetree and distribute it because this devicetree would be particular to your configuration and not applicable to the majority of our customers.

Regards,
Bogdan

adry84, 2020-04-24:
Hi Bogdan,
Thank you for your quick reply.

closed 2020-05-27 by acozma