analogdevicesinc/aditof_sdk #324: Segmentation fault in Open3d showPointCloud example

david-robbins, 2020-04-28:
When I try and run the Open3d showPointCloud example, I consistently get a segmentation fault on the following line in the getIntrinsic member function:

https://github.com/analogdevicesinc/aditof_sdk/blob/a08e42c54486266f09aa3712e7c31eb2098deac2/sdk/src/calibration.cpp#L196

It appears that data vector is corrupted before it is even passed into the function, which is a vector containing the camera matrix:

https://github.com/analogdevicesinc/aditof_sdk/blob/a08e42c54486266f09aa3712e7c31eb2098deac2/sdk/include/aditof/camera_definitions.h#L74

Any ideas on what's going on?

Thanks,
Dave R.

grigorovici-andreea, 2020-04-29:
Hello,

Can you tell if the other camera details from _aditof::CameraDetails cameraDetails_ are returned as expected? (Check the values for:
 _int camera_rangeMax = cameraDetails.maxDepth;
 int camera_rangeMin = cameraDetails.minDepth_)

Also, as a fast workaround, only to check the point cloud resulted, you could hard code the values for the intrinsic parameters: 
    double fx = 390.162;
    double fy = 385.983;
    double cx = 343.904;
    double cy = 234.95;

Thank you

david-robbins, 2020-04-29:
The camera details are being returned from the camera fine, they just can't be inserted into the corrupted cameraMatrix vector.

Here's the GDB output from when I tried to view the cameraMatrix vector:
[image]

I tried to comment out some lines to view the cameraDetails, but I continued to get segmentation faults in different places.  It seems like there's some form of widespread memory corruption.

dNechita, 2020-04-29:
Hi Dave,

I've seen this issue when compiling open3D from source and omitting to set this cmake option:
-DGLIBCXX_USE_CXX11_ABI=ON
How did you installed open3D?

david-robbins, 2020-04-29:
That was it.  I rebuilt Open3D with that cmake option and it worked perfectly.  Thanks for your help

closed 2020-04-29 by david-robbins