analogdevicesinc/aditof_sdk #337: Failed to get frame from device

adry84, 2020-04-30:
Hi all,

When I execute the imshow-jetson example,  
I have the following error : "select timeout"
Do you have an idea on how to resolved it?

Thanks

Here is the whole log :
./imshow-jetson 
I0430 15:47:38.585155 16031 device_enumerator_jetson.cpp:47] Looking for devices on the target
I0430 15:47:38.587298 16031 device_enumerator_jetson.cpp:80] Looking at: /dev/video0 for an eligible TOF camera
I0430 15:47:38.588415 16031 system_impl.cpp:62] System initialized
I0430 15:47:38.588513 16031 camera_96tof1.cpp:78] Initializing camera
I0430 15:47:38.588553 16031 local_device.cpp:158] Opening device
I0430 15:47:38.588582 16031 local_device.cpp:166] driverPath /dev/video0
I0430 15:47:38.588601 16031 local_device.cpp:167] driverPath /dev/video0
I0430 15:47:38.588658 16031 local_device.cpp:175] devName /dev/video0
I0430 15:47:38.588685 16031 local_device.cpp:176] subDevName  /dev/video0
I0430 15:47:38.858512 16031 calibration.cpp:261] EEPROM calibration data size 64940 bytes
I0430 15:47:40.971701 16031 camera_96tof1.cpp:95] Camera initialized
I0430 15:47:40.974328 16031 local_device.cpp:259] Starting device
I0430 15:47:40.974674 16031 camera_96tof1.cpp:129] Chosen mode: near
I0430 15:47:40.974726 16031 camera_96tof1.cpp:170] Camera range for mode: near is: 250 mm and 800 mm
I0430 15:47:40.975412 16031 camera_96tof1.cpp:180] Found firmware for mode: near
I0430 15:47:40.975446 16031 camera_96tof1.cpp:183] Firmware size: 10440 bytes
I0430 15:47:44.325201 16031 camera_96tof1.cpp:199] Camera calibration parameters for mode: near are gain: 1 offset: 0
W0430 15:48:05.950139 16031 local_device.cpp:869] select timeout
W0430 15:48:05.951397 16031 camera_96tof1.cpp:343] Failed to get frame from device
E0430 15:48:05.951632 16031 main.cpp:159] Could not request frame!
I0430 15:48:05.960408 16031 local_device.cpp:297] Stopping device

adry84, 2020-05-01:
Hi,

Today I try the aditof-demo example (on raspberry this time).
It gives me the following error :
```
Camera range for mode: near is: 250mm and 800 mm
Failed to read firmware eeprom 
terminate called after throwing an instance of ‘std::bad_array_ne_length’
	what(): std::bad_array_new_length
Aborted
```
Is my eeprom data corrupted?
If it is the case how to fix it?

Thanks.

btogorean, 2020-05-07:
Hi,

Please give us more details about your setup. You use the jetson image image we provide or you followed the steps presented on wiki?
Also what carrier model you have? A02 with one CSI port or B01 with two CSI ports?

Regards,
Bogdan

adry84, 2020-05-07:
Hi Bogdan, 

Thank you for your reply.

I use the master version of the sdk on jetson. 
I followed steps on your wiki.

I have also tested it on raspberry pi with the latest sdcard image. The log result is the same.

My AD-96TOF1-EBZ AFE Board is a rev C.
My jetson nano is a B01 with 2 CSI ports.

Regards.
Audrey

btogorean, 2020-05-08:
Looks like at this step the problem is not related specifically to Jetson or RPI.
Please check again the position of S1 and S5 required for ToF Rev.C camera presented [here](https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/ug_rpi)
Also after fresh boot please have a look at DMESG for suspicious prints coming from camera driver
`dmesg | grep addi`

Regards

adry84, 2020-05-11:
S1 is on position 1 
S5 : only S1 is to OFF

The output of dmesg | grep addi : 
[    6.467920] addi9036 0-0064: addi9036 detected at address 0x64

With dmseg --follow I have
...
   47.902774] addi9036 0-0064: could not write to register 0
[   47.902783] addi9036 0-0064: could not write to register 0
[   47.902788] addi9036 0-0064: could not write to register 0
[   47.902793] addi9036 0-0064: could not write to register 0
[   47.902797] addi9036 0-0064: could not write to register 0
[   47.902802] addi9036 0-0064: could not write to register 0
...
[image]

btogorean, 2020-05-14:
Hi,

The switches setup look ok for RPi configuration.

The "could not write to register 0" is expected for now. It happens because we define the following [regmap](https://github.com/analogdevicesinc/linux/blob/b7ae5208539373c05969db561664d9947a62ec48/drivers/media/i2c/addi9036.c#L140) writable registers but on bulk write of ADDI9036 firmware we have transmissions that are interpreted as write to address 0.

The problem should be in other part but is not clear to me yet where.
I would like to ask you if the camera worked before or you never were able to start it?
I have to check with the rest of the team the possibility of rewriting the EEPROM but the main problem with this procedure is that EEPROM contains some calibration data unique for each specific camera.

Regards,
Bogdan

adry84, 2020-05-14:
Hi Borgan,
Thank you for your reply.
The camera used to work fine. 
I used it with the jetson only. 
I modified the device tree to be able to use both camera ports on the jetson board so that I can use at the sae time a standard color rpi camera and your tof camera.
Since then I have not been able to make it work. I rollbacked to the original device tree on jetson nano and I tested it on raspberry without success.

adry84, 2020-05-14:
If that can help : I retrieved the content of the eeprom.
[eeprom_data_hexdump.txt](https://github.com/analogdevicesinc/aditof_sdk/files/4628841/eeprom_data_hexdump.txt)

littleyueh, 2020-06-13:
Hi @adry84 

I face some similar questions just like yours
May I ask do you have any solution?

this is my issue link:
https://github.com/analogdevicesinc/aditof_sdk/issues/373

thanks

acozma, 2020-07-16:
The issue was related to a hardware malfunction.

closed 2020-07-16 by acozma