analogdevicesinc/aditof_sdk #373: Failed to get frame (RPi)

littleyueh, 2020-06-10:
Hi, 

I have already built the SDK by installing the image file on my Rpi 4
[image]

and I follow the startup settings from this link: https://wiki.analog.com/resources/eval/user-guides/ad-96tof1-ebz/ug_rpi#troubleshooting
(I choose AD-96TOF1-EBZ rev.C, and I also turned the S1 on, S5 switch 1 off and others on)

However, it returns "Failed to get frame" 
the following messages are from terminal

I0610 08:28:17.203526 14105 device_enumerator_raspberry.cpp:47] Looking for devices on the target
I0610 08:28:17.204475 14105 device_enumerator_raspberry.cpp:80] Looking at: /dev/video0 for an eligible TOF camera
I0610 08:28:17.205425 14105 system_impl.cpp:62] System initialized
I0610 08:28:17.205514 14105 camera_96tof1.cpp:78] Initializing camera
I0610 08:28:17.205540 14105 local_device.cpp:156] Opening device
I0610 08:28:17.982082 14105 calibration_96tof1.cpp:163] EEPROM calibration data size 64940 bytes
I0610 08:28:23.740208 14105 camera_96tof1.cpp:95] Camera initialized
I0610 08:28:23.742771 14105 local_device.cpp:251] Starting device
I0610 08:28:23.776517 14105 camera_96tof1.cpp:129] Chosen mode: near
I0610 08:28:23.776710 14105 camera_96tof1.cpp:170] Camera range for mode: near is: 250 mm and 800 mm
I0610 08:28:23.777487 14105 camera_96tof1.cpp:180] Found firmware for mode: near
I0610 08:28:23.777516 14105 camera_96tof1.cpp:183] Firmware size: 10440 bytes
I0610 08:28:25.057718 14105 calibration_96tof1.cpp:376] Camera calibration parameters for mode: near are gain: 1 offset: 0
I0610 08:28:25.057930 14105 calibration_96tof1.cpp:388] Camera intrinsic parameters:
    fx: 395.533
    fy: 392.584
    cx: 327.293
    cy: 237.566
W0610 08:28:33.739828 14111 local_device.cpp:863] select timeout
W0610 08:28:33.740283 14111 camera_96tof1.cpp:316] Failed to get frame from device

I will appreciate any help and advice, thanks

dNechita, 2020-06-10:
Hi,

Can you open another terminal, run
dmesg -W
and see what is being printed on the terminal while you start the sdk example? And share it with us?

littleyueh, 2020-06-10:
Sure,

The results show that it could not write to register 0

[image]

dNechita, 2020-06-10:
Can you send the entire log?
(After you get the timeout, run
dmesg > my_log.txt
to get the entire output and store it into a file)

littleyueh, 2020-06-11:
Sure, the following is the entire log:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.86-v7l+ (btogorean@btogorean-pc) (gcc version 9.1.0 (GCC)) #1 SMP Tue Dec 3 17:21:04 EET 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 999424
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 802816 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4ec with crng_init=0
[    0.000000] percpu: Embedded 16 pages/cpu s35404 r8192 d21940 u65536
[    0.000000] pcpu-alloc: s35404 r8192 d21940 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 997696
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:45:0C:1E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d9b3f436-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3680664K/3997696K available (10240K kernel code, 774K rwdata, 3324K rodata, 2048K init, 853K bss, 54888K reserved, 262144K cma-reserved, 3211264K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (12256 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 775 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 854 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31282 entries in 92 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000235] Console: colour dummy device 80x30
[    0.000259] console [tty1] enabled
[    0.000302] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000323] pid_max: default: 32768 minimum: 301
[    0.000579] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000596] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001339] CPU: Testing write buffer coherency: ok
[    0.001737] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002354] Setting up static identity map for 0x200000 - 0x20003c
[    0.002506] rcu: Hierarchical SRCU implementation.
[    0.003339] smp: Bringing up secondary CPUs ...
[    0.004158] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005107] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006015] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006148] smp: Brought up 1 node, 4 CPUs
[    0.006161] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.006170] CPU: All CPU(s) started in HYP mode.
[    0.006178] CPU: Virtualization extensions available.
[    0.006929] devtmpfs: initialized
[    0.017596] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.017804] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017824] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.024297] pinctrl core: initialized pinctrl subsystem
[    0.025127] NET: Registered protocol family 16
[    0.027846] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.029279] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... 748 lines cut ...]
[  142.919378] addi9036 0-0064: could not write to register 0
[  142.919382] addi9036 0-0064: could not write to register 0
[  142.919387] addi9036 0-0064: could not write to register 0
[  142.919391] addi9036 0-0064: could not write to register 0
[  142.919396] addi9036 0-0064: could not write to register 0
[  142.919400] addi9036 0-0064: could not write to register 0
[... cut ...]
[  142.919449] addi9036 0-0064: could not write to register 0
[  142.919453] addi9036 0-0064: could not write to register 0
[  142.919457] addi9036 0-0064: could not write to register 0
[  142.919462] addi9036 0-0064: could not write to register 0
[  142.919466] addi9036 0-0064: could not write to register 0
[  142.919471] addi9036 0-0064: could not write to register 0
[  142.919475] addi9036 0-0064: could not write to register 0
[  142.919480] addi9036 0-0064: could not write to register 0
[  142.919484] addi9036 0-0064: could not write to register 0

littleyueh, 2020-06-12:
Hi @dNechita 

cuz I don't know how to fix these problems
All I can try is to rebuild the SD card image and change the system to Rpi 3

but it still returned same error messages
sincerely hope that there is any help and advise, thx

dNechita, 2020-06-12:
The message: " addi9036 0-0064: could not write to register 0" can be considered a warning. These messages are not the reason your are not getting any frames.
This may sound like a silly request, but please check again the position of all switches and make sure that the 15 pin flex cable is connected properly.
Please let me know if the problem persists.

littleyueh, 2020-06-12:
Hi,

I have tried to reconnect the 15 pin flex cable, and make sure the cable is connected properly
and I also checked each pin (by multimeter), all of them are no broken

the switches are set as follow
[image]

thanks

dNechita, 2020-06-12:
The position of the switches looks good. Thanks for posting them.
After you press the 'Play' button from the demo, can you check if leds DS2 and DS3 are turned on?

littleyueh, 2020-06-12:
yes both of them are on

[image]

dNechita, 2020-06-12:
That's good, it means that the device is successfully programed.
Not sure why you don't get any frames. We have tested with the latest image and we can't reproduce.
Did your ad-96tof1-ebz board worked before?

littleyueh, 2020-06-12:
Yes, it has worked when I first time turn on 
then I cannot get the frame at the next time, I didn't change any configuration
so it really confused me a lot

littleyueh, 2020-06-13:
Hi @dNechita 

I have tested the CSI port on my RPi 3, with the RPi camera
it still works fine

So far I've tried many possible solutions
but I still cannot figure out the reason

btogorean, 2020-06-15:
Hi @littleyueh 

From the SDK error it looks like the communication on CSI clock or data lines is not executed properly.

Do you have the possibility to check the CSI data and clock differential lines with an oscilloscope and see if there is any activity after you press play?

If you don't have a scope press play and check if you can hear something like a high frequency noise coming from camera that should change frequency or intensity if you move your hand closer or further from camera?

littleyueh, 2020-06-15:
Hi @btogorean 

about the so-called "high-frequency" noise, my experience is that when I change the distance mode on SDK (from near to far), the noise's frequency will sound more higher
meanwhile, the LED DS2 will turn off on the transient tim

and I did the experiment that you advised, the result is that when I move my hand closer, the camera module will have a lower frequency sound

tomorrow I can use the oscilloscope to check the clock and data
cuz I am not familiar with CSI protocol, I want to ask which pin should I test first? SCL0 and SDA0?

fyi, this is the rpi3 CSI port schemetic:
[image]
(https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_3b_1p2_reduced.pdf)

thanks for your reply

littleyueh, 2020-06-15:
Hi @btogorean 

I just checked the CSI pins, and the results are shown as following pictures,

CAM1_DN0, CAM1_DP0 (differential camera data):
[image]

CAM1_DN1, CAM1_DP1 are quite similar with the first differential camera data

CAM1_CN, CAM1_CP: nothing, both of them are 0 volts

SCL0, SDA0: 3.3V, I think the I2C communication didn't start, so both of them are pulled to high level

It seems that the problem is clock issue, hope these results can help 

thanks a lot

btogorean, 2020-06-16:
Thank you for oscilloscope prints

Good news is that the camera is putting out data and timing wise looks ok.

One observation is that the positive and negative signals of one differential pair (for ex. CAM1_DN0, CAM1_DP0) should have the same amplitude.
In your case it looks like the red one has much smaller amplitude. Please check that your probe is not set on X10 or the electric contact is good.

Also on clock line there should be a clock of 135Mhz. Check that your oscilloscope has enough bandwidth to measure such high frequency signals. If it really is missing that's a problem.

Other than that please post the output of `v4l2-ctl --all` and `media-ctl -p` console commands .

P.S. For Rpi3 it should be `media-ctl -d /dev/media1 -p`

littleyueh, 2020-06-16:
Hi @btogorean , thanks for your reply

1. Yes, yesterday my experiment was wrong, I didn't check my probe, one of them is set on X10
the following is the correct measurement, all probes are set on X1, and both's amplitude are similar

 
[image]

2. About the clock, I measured it by logic analyzer
[image]
it shows that CAM1_CN and CAM1_CP are in low-level, no clock generated

fyi, these are the results of measuring data line and power line (pin 15), it shows data line has rising/falling edge and power line has a constant high level signal

[image]

[image]

3. I test my 96TOF on RPi-3
this is the console log after typing `v4l2-ctl --all`
[image]

and this is the console log after typing `media-ctl -d /dev/media1 -p`
[image]

hope these information will help
thanks a lot

btogorean, 2020-06-17:
Hi

1. Looks ok. It can be seen also the Start of Frame sequence

2. It really depends also on your LA front-end bandwidth. For example on my 200Mhz bandwidth oscilloscope I can see the 135MHz clock but at an amplitude of ~ 50mV. This clock is generated by camera. If it is not there it really makes sense to get the error you're having. I simulated by disconnecting the cable between camera and RPi and the error was the same as yours. Please change the flex cable if is possible or measure on the camera side, P1 connector.

3. The video pipeline looks as it should so I think that SW wise we are ok.

littleyueh, 2020-06-17:
Hi @btogorean 

Thanks for your reply.
About the LA front-end setting, I set the sample rate at 1GHz, and when the input signal is bigger than 0.75V, it will return logical high. 

I have changed new 15-pin flex cable, and measured on the P1 connector
Unfortunately, it turns out no signal on the differential clock pins.

btogorean, 2020-06-17:
Hi @littleyueh 

You said before that you have a RPi camera.
Can you please try with that to check if using that there is activity on differential clock lines?

If we don't have clock for sure image capture will not work

Thank you

littleyueh, 2020-06-18:
Hi @btogorean 
thanks for your patience and reply

I have tried the my RPi camer with RPi3 (with ADI ToF image)
then it returns as following

[image]

Then I tried official image from Raspberry on RPi3
it can capture the video, also the differential clock lines have signals
The command I typed is  `raspivid -t 50000`
(LA setting: 1GHz, upper 0.75V returns logic high)

[image]

Is there any possibility that the clock issue is happened on the 96TOF board?
If so, how do I fix this problem?

Thank you

btogorean, 2020-06-18:
I think if you comment `dtoverlay=rpi-addi9036,revc` on boot partition in config.txt file the ADI ToF image should work with RPi camera.

I have to check if there is possible to happen something with clock output of 96TOF board

littleyueh, 2020-06-18:
Hi @btogorean 

yes, I followed your tip and changed the config file
now I can get the video from RPi camera with ADI ToF image file
[image]

and I measure the clock pin, it has signals
(LA configuration is as usual)

[image]

thanks

littleyueh, 2020-06-20:
Hi @btogorean 

It seems that the issue is happened on my 96TOF board
I set the configuration as same RPi, same image file and same 15-pin flex cable, but different camera modules---RPi camera and 96TOF, only 96TOF's differential clock pins have no signals.

Is there any solution that can fix the problems?
Really appreciate your help and any advice.

Thanks

btogorean, 2020-06-23:
Hi,

Sorry for late response. I would like to ask you to create a dump of your EEPROM firmware and share it with us to be validated. There are some registers that alter the clock output and if the firmware got corrupted maybe are not correctly set.

You can use the command:
`sudo dd if=/sys/bus/i2c/devices/0-0056/eeprom of=dump.bin`

littleyueh, 2020-06-24:
Hi, @btogorean 

thanks for your reply
here is the firmware .bin file

[dump.zip](https://github.com/analogdevicesinc/aditof_sdk/files/4823621/dump.zip)

please check, thanks for your help

btogorean, 2020-06-24:
I tried the firmware on my ToF camera + Rpi and it is working as expected.

I would like to ask you if by any chance you have a Dragonboard 410c and can give it a try on that platform?

littleyueh, 2020-06-24:
Hi,

thanks for your testing
Unfortunately, I don't have a Dragonboard 410c  to test Q____Q

It's weird and confuse me a lot, but I will try to rebuild the image file and double check all my configuration

bitslip6, 2020-06-25:
I have been following this thread for a little while.  We are also running into the same issue but on IMX8 board with our custom AFE  board.

We have also verified firmware is running correctly on PI.  My scope here is only 100Mhz so difficult to really measure the CSI interface.  Our issue may be device tree related where the pin mappings are not as expected (all custom hardware).   I am going to attempt to get data from a pi camera here and see if that can rule out the device tree issue.  I'll post any results we have on this similar issue on this thread as well.

littleyueh, 2020-06-26:
Hi @bitslip6 

Thanks for your effort
Really look forward to any results

KamelAbdelouahab, 2020-06-29:
Hi all,

We're facing the same problem on Raspberry Pi 3 and Jetson Nano using the 96-TOF1-EBZ board. I also noticed a noisy sound from the board that varies according to the sensor range. 
The debug provided in this thread is really helpful. Looking forward for more results.

acozma, 2020-06-29:
Hi @littleyueh 

This really looks like a hardware failure on the 96ToF1 board, especially considering the board worked once. Can you do a visual inspection of the board on the bottom side around U5, U7 and U11 to see if there isn't anything suspicions on the PCB?

littleyueh, 2020-06-29:
Hi @acozma 

thanks for your reply
I have visually checked the PCB board

I found:

1. around U5: R60, R55, D2, R63, R54 have no components on these locations
2. around U7: nothing suspicious
3. around U11: nothing suspicious

For your further information, due to ESD issue, every time I use TOF module, I carefully avoid touching the components on PCB board and I also wear anti-static wrist trap.

hope these information will help

acozma, 2020-06-29:
Hi @littleyueh

Can you probe the clock lines directly on pins 1&2 of U7? This will tell is there's something wrong with the ADDI9036 or with any of the MIPI switches (U7 & U11).

littleyueh, 2020-06-29:
Hi @acozma 

Sure, I measured it by LA

Myy probe is on these two pins shown in the picture, and GND is on RPi's GND pin
[image]

I probed Pin1&2 on U7
However, there is no signal on both pins (pin3&4 have signals)

cuz I need to hold the probe and ToF module, so I didn't take the picture of the results on LA, sorry about that

bitslip6, 2020-06-29:
> Hi all,
> 
> We're facing the same problem on Raspberry Pi 3 and Jetson Nano using the 96-TOF1-EBZ board. I also noticed a noisy sound from the board that varies according to the sensor range.
> The debug provided in this thread is really helpful. Looking forward for more results.

We are also investigating this issue.  Our current theory is expansion and contraction of the 100uF MLCC components on the illumination board.  We are going to try a design with 3 47Uf 0805 caps to see if that reduces noise.  Our second theory is the FETs on the illumination board.

acozma, 2020-06-29:
Hi @littleyueh, @bitslip6,

It might be worth probing the LDx and TALx signals on the laser board connector to see the offset between each LDx & TALx pair. If the TALx signal does not have a relatively constant offset to the LDx signal or the waveforms are distorted that will cause the ADDI9036 to not output valid data on the MIPI lines. The TAL signal will also tell if the FET is working as expected.

littleyueh, 2020-07-07:
Hi @acozma 

Sorry for late reply
I am wondering where are LDx & TALx pins, because I cannot find both of them on the laser board.
Could you point out where they are? Thank you.

acozma, 2020-07-07:
There are 4 pairs, LD1&TAL1, LD2&TAL2, ... All are coming to the connector between the laser and the camera board so it might be easier to probe somewhere around the connector. Without soldering some wires probing might be difficult since all the signals are routed on the bottom of the laser board and the top of the camera board overlapping with the laser board.

closed 2021-10-29 by acozma