analogdevicesinc/aditof_sdk #387: Problems with imshow example compiling

littleyueh, 2020-07-15:
Hi, 

I want to add some extra functions on the /bindings/opencv/imshow/main.cpp, and compile it to a .exe file for me
So I download the image file (for Rpi system), copy the /workplace/aditof_sdk to desktop, and rewrite the main.cpp

then I go to /Desktop/aditof_sdk to run the cmake command, want to build a .exe file just like /build/bindings/opencv/imshow/aditof-opencv-show

but it returns the following error message: Could not found protobuf
[image]

I saw the document said the image file has all the dependencies package.
Should I re-download the protobuf package again?

I will appreciate any help and advice, thanks

littleyueh, 2020-07-16:
Hi,

I've changed my cmake command and the following is the result:
[image]

The configuration is done, but when I tried to build, it stuck on the step of building "device_enumerator_ethernet.cpp.o"
I checked the /aditof_sdk/sdk/src/connections/network, there is no such header file called "buffer.pb.h"

I will appreciate any help and advice, thanks

dNechita, 2020-07-16:
Hi,

We had this fixed recently.
Can you try with the latest master branch?

littleyueh, 2020-07-16:
Hi @dNechita 

I got it
It works, thanks!

dNechita, 2020-07-16:
Glad to hear that!
If you are disabling the network support then you no longer need the following dependecies: websockets and protobuf (no need to add their paths to CMAKE_PREFIX_PATH)

Closing this.

closed 2020-07-16 by dNechita