analogdevicesinc/aditof_sdk #388: Error while installing the SDK

sindhuvahinis, 2020-08-03:
While trying to install the sdk using cmake --build . --target install

I am getting the below error. 

glog.lib(logging.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(logging.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(utilities.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(utilities.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(raw_logging.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(raw_logging.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(vlog_is_on.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(vlog_is_on.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(port.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]
glog.lib(port.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in camera_factory.obj [\aditof_sdk\build\sdk\aditof.vcxproj]

\aditof_sdk\build\sdk\Debug\aditof.dll : fatal error LNK1319: 10 mismatches detected [\aditof_sdk\build\sdk\aditof.vcxproj]

Am I missing something?

dNechita, 2020-08-04:
This happens due to the fact that glog.lib has been built for release while you are building the aditof_sdk in debug mode. How did you run cmake (can you share the exact line)?

Another way around this is to build aditof_sdk in release mode:
`cmake --build . --config "Release" --target install`

sindhuvahinis, 2020-08-04:
Your suggestion worked. Thank you very much!

closed 2020-08-04 by sindhuvahinis