analogdevicesinc/aditof_sdk #423: Continuous Integration enhancements

DanielGuramulta, 2020-11-11:
1. Some pull requests might not touch any source files (readme updates, python files that are not included in ci, scripts that are not included in the ci). For this builds we can check that no source file (.cpp .h, cmake files) are modified and immediately return success.

2. Does it make sense to run the ci only on pull requests and master by default. And have a naming convention for branches that wish to run ci jobs on them that are not yet a pull request. ex: fix-demo-bug+ci. We can match for branches that have `+ci` at the end.

VLitanADI, 2020-11-12:
1. great idea. I would suggest to also have the option to run only the formatting checks for some files (ex python scripts)

DanielGuramulta, 2020-11-12:
From what I remember we only run formatting on c++ files (clang-format). We could also add formatting to python files, shell scripts, cmake files.
1. For cmake we can use cmake-format which can also do some static analysis (https://github.com/cheshirekow/cmake_format)
2. For shell scripts I don't know any formatting tool, but there is shell check that does static analysis (https://www.shellcheck.net)
3. python: https://github.com/google/yapf

closed 2021-10-29 by acozma