analogdevicesinc/aditof_sdk #483: Migrate Travis CI to Azure CI

dNechita, 2021-01-07:


VLitanADI, 2021-01-28:
Should always build with:

- [x] Examples
- [x] Network
- [x] Python

Flag combinations (will run on ubuntu18 if no DB or RP is specified) :
- [ ] 96TOF1 + DGB410c
- [ ] 96TOF1 + RPI
- [ ] 96TOF1 + Jetson
- [ ] 96TOF1 + Xavier NX
- [ ] 96TOF1 + Xavier AGX
- [ ] FX1 + Toybrick
- [ ] FX1 + RPI
- [ ] FX1 + Jetson
- [ ] FX1 + Xavier NX
- [ ] FX1 + Xavier AGX

Platforms to build on (with no other flags):

- [ ] Ubuntu 18, clang7
- [ ] Ubuntu 18, clang8
- [ ] Ubuntu 18, gcc7
- [ ] Ubuntu 18, gcc8
- [ ] Ubuntu 20, clang7
- [ ] Ubuntu 20, clang8
- [ ] Ubuntu 20, gcc7
- [ ] Ubuntu 20, gcc8
- [x] Raspberry
- [x] Dragonboard

Code quality and doc:
- [ ] clang_format
- [ ] cppcheck
- [ ] deploy doc

VLitanADI, 2021-01-28:
- [ ] Rename folder with scripts to "Azure"
- [ ] Remove unused scripts
- [ ] Remove travis.yaml

closed 2021-02-04 by VLitanADI