analogdevicesinc/aditof_sdk #547: XavierNX : Permission denied and missing udev ?

KamelAbdelouahab, 2021-04-09:
Hi all,

I'm experimenting with the addi9036 camera and a Jetson XavierNX. I built the device tree, kernel, modules and  SDK according to the guides and I came to the first-frame example. 

```
I20210409 18:41:30.203469 30733 sensor_enumerator_xaviernx.cpp:49] Looking for devices on the target: Xavier NX
I20210409 18:41:30.209442 30733 camera_96tof1.cpp:146] Initializing camera
I20210409 18:41:30.209720 30733 addi9036_sensor.cpp:140] Opening device
I20210409 18:41:30.209834 30733 addi9036_sensor.cpp:158] Looking for the following cards:
I20210409 18:41:30.209911 30733 addi9036_sensor.cpp:160] vi-output, addi9036 9-0064
I20210409 18:41:30.210065 30733 addi9036_sensor.cpp:160] vi-output, addi9036 10-0064
E20210409 18:41:30.227255 30733 eeprom.cpp:64] fopen() failed. Error: Permission denied
E20210409 18:41:30.230532 30733 camera_96tof1.cpp:172] Failed to open EEPROM with name 24c1024
E20210409 18:41:30.230655 30733 main.cpp:59] Could not initialize camera!
```

I thought it was a udev permission problem, I looked for an adequate udev file similar to the AGX Xavier and Jetson but did not find one. So I authorized the following devices: 
```
SUBSYSTEM=="i2c", 
RUN+="/bin/chown root:plugdev /sys/bus/i2c/devices/9-0056/eeprom",
RUN+="/bin/chmod 0060 /sys/bus/i2c/devices/9-0056/eeprom",
RUN+="/bin/chown root:plugdev /sys/bus/i2c/devices/9-0064/eeprom",
RUN+="/bin/chmod 0060 /sys/bus/i2c/devices/9-0064/eeprom",
RUN+="/bin/chown root:plugdev /sys/bus/i2c/devices/10-0064/eeprom",
RUN+="/bin/chmod 0060 /sys/bus/i2c/devices/10-0064/eeprom",
RUN+="/bin/chmod 666 /dev/i2c-9"
```

The `9-0056` came from the source code while the `9-0064` and `10-0064` came from the logs, but still the same problem. Does this come from my udev rules or something else ? I also have 

If it helps, I also give you the following dmesgs: 
```
dmesg | grep addi
[    2.027370] addi9036 9-0064: probing addi9036 v4l2 sensor
[    2.027419] addi9036 9-0064: reset-gpios not found -2
[    2.027847] addi9036 9-0064: Detected ADDI9036 sensor
[    2.027884] addi9036 10-0064: probing addi9036 v4l2 sensor
[    2.027933] addi9036 10-0064: reset-gpios not found -2
[    2.028132] addi9036 10-0064: couldn't create debugfs
[    2.028367] addi9036 10-0064: Detected ADDI9036 sensor
[    3.023453] tegra194-vi5 15c10000.vi: subdev addi9036 9-0064 bound
[    3.024766] tegra194-vi5 15c10000.vi: subdev addi9036 10-0064 bound
```

```
dmesg | grep i2c
[    0.732871] iommu: Adding device 3160000.i2c to group 10
[    0.733437] iommu: Adding device c240000.i2c to group 11
[    0.733836] iommu: Adding device 3180000.i2c to group 12
[    0.734240] iommu: Adding device 3190000.i2c to group 13
[    0.734856] iommu: Adding device 31b0000.i2c to group 14
[    0.735242] iommu: Adding device 31c0000.i2c to group 15
[    0.735612] iommu: Adding device c250000.i2c to group 16
[    0.736049] iommu: Adding device 31e0000.i2c to group 17
[    0.860910] tegra-i2c 31e0000.i2c: could not find pctldev for node /host1x/dpaux@155F0000/pinmux@0, deferring probe
[    1.996791] i2c /dev entries driver
[    1.999661] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.001405] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.003142] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.004862] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.006561] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.008247] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.009907] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.011560] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.013208] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.014865] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.016526] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.017906] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.019558] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.021239] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.022932] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.024619] tegra-i2c 3180000.i2c: no acknowledge from address 0x56
[    2.026002] i2c i2c-2: Added multiplexed i2c bus 9
[    2.027200] i2c i2c-2: Added multiplexed i2c bus 10
[    2.027208] i2c-mux-gpio cam_i2cmux: 2 port mux on 3180000.i2c adapter
[   14.265212] tegra-i2c 3180000.i2c: no acknowledge from address 0x64
[   14.280816] tegra-i2c 3180000.i2c: no acknowledge from address 0x64
[  223.633517] tegra-i2c 3180000.i2c: no acknowledge from address 0x64
[  223.640058] tegra-i2c 3180000.i2c: no acknowledge from address 0x64
[  599.398251] tegra-i2c 3180000.i2c: no acknowledge from address 0x64
[  599.415172] tegra-i2c 3180000.i2c: no acknowledge from address 0x64
```

Thanks,
K

D3-aavery, 2021-04-13:
I am having this same issue. Note however that it only happens on Rev C, and not Rev B boards.

dNechita, 2021-04-14:
Hi,

I'm not sure why things are not working. But before all things, have you tried to use the existing SD card image? It can be found on the main page (https://github.com/analogdevicesinc/aditof_sdk) in the AD-96TOF1-EBZ section.

VLitanADI, 2021-04-14:
Hi, 
Please run 
`media-ctl --print-topology` 
or
`media-ctl --print-dot > graph.dot`
`dot -Tpng graph.dot > graph.png`
and post the output

KamelAbdelouahab, 2021-04-14:
> Hi,
> Please run
> `media-ctl --print-topology`
> or
> `media-ctl --print-dot > graph.dot`
> `dot -Tpng graph.dot > graph.png`
> and post the output

Here is the output graph: 
[image]
```
Media controller API version 0.1.0

Media device information
------------------------
driver          tegra194-vi5
model           NVIDIA Tegra Video Input Device
serial          
bus info        
hw revision     0x3
driver version  0.0.0

Device topology
- entity 1: addi9036 9-0064 (1 pad, 1 link)
            type V4L2 subdev subtype Sensor flags 0
            device node name /dev/v4l-subdev0
	pad0: Source
		[fmt:SRGGB12_1X12/640x480 field:none colorspace:srgb]
		-> "15a00000.nvcsi--2":0 [ENABLED]

- entity 3: 15a00000.nvcsi--2 (2 pads, 2 links)
            type V4L2 subdev subtype Unknown flags 0
            device node name /dev/v4l-subdev1
	pad0: Sink
		<- "addi9036 9-0064":0 [ENABLED]
	pad1: Source
		-> "vi-output, addi9036 9-0064":0 [ENABLED]

- entity 6: vi-output, addi9036 9-0064 (1 pad, 1 link)
            type Node subtype V4L flags 0
            device node name /dev/video0
	pad0: Sink
		<- "15a00000.nvcsi--2":1 [ENABLED]

- entity 18: addi9036 10-0064 (1 pad, 1 link)
             type V4L2 subdev subtype Sensor flags 0
             device node name /dev/v4l-subdev2
	pad0: Source
		[fmt:SRGGB12_1X12/640x480 field:none colorspace:srgb]
		-> "15a00000.nvcsi--1":0 [ENABLED]

- entity 20: 15a00000.nvcsi--1 (2 pads, 2 links)
             type V4L2 subdev subtype Unknown flags 0
             device node name /dev/v4l-subdev3
	pad0: Sink
		<- "addi9036 10-0064":0 [ENABLED]
	pad1: Source
		-> "vi-output, addi9036 10-0064":0 [ENABLED]

- entity 23: vi-output, addi9036 10-0064 (1 pad, 1 link)
             type Node subtype V4L flags 0
             device node name /dev/video1
	pad0: Sink
		<- "15a00000.nvcsi--1":1 [ENABLED]
```
@dNechita I'm currently downloading the image. Updates incoming ..

KamelAbdelouahab, 2021-04-14:
I'm having the same issue with the provided SD card image for the XavierNX.
Here are the logs of the aditof-demo
```
I0128 18:01:25.472430  7369 sensor_enumerator_xaviernx.cpp:49] Looking for devices on the target: Xavier NX
I0128 18:01:25.480610  7369 camera_96tof1.cpp:144] Initializing camera
I0128 18:01:25.480767  7369 addi9036_sensor.cpp:139] Opening device
I0128 18:01:25.480849  7369 addi9036_sensor.cpp:157] Looking for the following cards:
I0128 18:01:25.481045  7369 addi9036_sensor.cpp:159] vi-output, addi9036 9-0064
I0128 18:01:25.481082  7369 addi9036_sensor.cpp:159] vi-output, addi9036 10-0064
E0128 18:01:25.487833  7369 eeprom.cpp:64] fopen() failed. Error: Permission denied
E0128 18:01:25.488654  7369 camera_96tof1.cpp:170] Failed to open EEPROM with name 24c1024
I0128 18:01:25.491360  7369 addi9036_sensor.cpp:254] Starting device 0
I0128 18:01:25.497766  7369 addi9036_sensor.cpp:254] Starting device 1
Gtk-Message: 18:01:25.591: Failed to load module "canberra-gtk-module"
I0128 18:01:25.602912  7369 camera_96tof1.cpp:231] Chosen mode: near
I0128 18:01:25.603083  7369 camera_96tof1.cpp:273] Camera range for mode: near is: 250 mm and 800 mm
W0128 18:01:25.603145  7369 camera_96tof1.cpp:280] Failed to read firmware from eeprom
```

Note that my AD-96TOF1-EBZ worked just fine with an RPIv3b and an RPIv4.

dNechita, 2021-04-15:
Can you try running aditof-demo with sudo?

KamelAbdelouahab, 2021-04-15:
> Can you try running aditof-demo with sudo?

Yes I tried, and the output was the same...

btw, is this normal at boot time ? 
```
tegra-i2c 3180000.i2c: no acknowledge from address 0x56
```

VLitanADI, 2021-04-16:
> btw, is this normal at boot time ?
> 
> ```
> tegra-i2c 3180000.i2c: no acknowledge from address 0x56
> ```
Not really, that is the address of the EEPROM.

VLitanADI, 2021-04-16:
Are you connecting the camera to CAM1 connector of the board?

KamelAbdelouahab, 2021-04-16:
> Are you connecting the camera to CAM1 connector of the board?

Nop ! The board is conencted to the CAM 0 port. I can try on CAM 1

D3-aavery, 2021-04-16:
I'm using CAM1 and still see the issue on my Rev C dev kit. I do not see the issue with Rev B or D dev kits.

KamelAbdelouahab, 2021-04-26:
Update: The following tests have been performed on a Rev C board with the images downloaded from this repo and a FFC cable provided with the board: 
- XavierNX CAM0    -> Fail
- XavierNX CAM1    -> Fail
- RPv3                     -> Sucess
- RPv4                     -> Sucess
- Jetson Nano A02   -> The imshow-jetson code works, but other implementations (python sdk, demo app) fail ...

KamelAbdelouahab, 2021-04-26:
BTW: Here is how the captured frames look like on a Jetson Nano (DepthMap above, NIR below)

https://user-images.githubusercontent.com/17831821/116100234-06e29e00-a6ad-11eb-8456-5d320aabf43d.mp4

VLitanADI, 2021-04-27:
Hi,
Regarding Jetson Nano A02:

- Demo app is not supposed to work on this platform. Due to some hardware limitations this board is not able to capture at the same time both Depth and IR frames. This is why a dedicated example exists. 
- Could you describe how the python SDK fails?

KamelAbdelouahab, 2021-04-27:
Hello, 

Thanks for your clarifications. My python SDK based script failed because I requested `depth_ir` frame types, resulting in this kind of messy frames: 
[image]

Note that I all manipulations on the Nano fails until I run the `first-frame` script
```
./imshow-jetson
...
W0426 17:20:28.482671  8671 camera_96tof1.cpp:280] Failed to read firmware from eeprom
...
./first-frame
...
I0426 17:20:42.856901  8678 camera_96tof1.cpp:286] Firmware size: 10440 bytes
...
./imshow-jetson
...
W0426 17:22:28.856901  8678 camera_96tof1.cpp:286] Firmware size: 10440 bytes
...

KamelAbdelouahab, 2021-05-24:
I'm still facing the issue with XavierNX boards (tested on another devkit today), and I can't get my hand on a Rev D board ...
Do you think I can patch 96TOF1 Rev-C board on my own ? What's the main differences between Rev-C and Rev-D ?

closed 2021-06-30 by KamelAbdelouahab

gianlucapargaetzi, 2023-04-03:
Hi, I‘m currently having the same Problem with my ADDI9036 (FXTOF) camera on a Xavier NX.

How was this issue resolved?