analogdevicesinc/aditof_sdk #562: Can't capture frames with AD-FXTOF1-EBZ

KamelAbdelouahab, 2021-05-25:
Hi all, 

We recently had the opportunity to evaluate the AD-FXTOF1-EBZ camera. Unfortunately, all our attempts to capture frames have failed. 

Our tests have been performed on Raspberry Pis and XavierNX boards. We followed the wiki instruction and used the provided Linux images. In all the cases (RPI3b, RPI4, XavierNX devkit), the camera was successfully detected: 
```
[3.900207] addi9036 0-0064: addi9046 detected at address 0x64
```

When running the demo script, or even the first-frame script, the camera calibrates then everything crashes 
```
[390.386905] addi9036 0-0064: could not write to register 0
```
In fact, the problem is quite similar to the one discussed  [here](https://github.com/analogdevicesinc/aditof_sdk/issues/373)

Any ideas why is this happening ? 

Thanks for the assistance,
Kamel

PS1: We already used the correct Flex cable and not the one provided in the package
PS2: We used the correct linux images for the FXTOF and not the 96TOF1

SeptimiuVana, 2021-05-27:
Hello Kame,

Can you please send us the log from aditof-demo?

KamelAbdelouahab, 2021-05-27:
Hi, 

Here is the adi tof demo logs for a raspberry pi 3B: 

```
I0527 16:24:35.618257   999 sensor_enumerator_raspberrypi.cpp:49] Looking for sensors on the target: Raspberry PI
I0527 16:24:35.623595   999 camera_fxtof1.cpp:127] Initializing camera
I0527 16:24:35.623755   999 addi9036_sensor.cpp:140] Opening device
I0527 16:24:35.623906   999 addi9036_sensor.cpp:158] Looking for the following cards:
I0527 16:24:35.623982   999 addi9036_sensor.cpp:160] unicam
I0527 16:24:36.460032   999 camera_fxtof1.cpp:178] Camera initialized
I0527 16:24:36.466727   999 addi9036_sensor.cpp:257] Starting device 0
I0527 16:24:36.586297   999 camera_fxtof1.cpp:209] Chosen mode: near
I0527 16:24:36.586474   999 camera_fxtof1.cpp:222] Camera range for mode: near is: 250 mm and 800 mm
I0527 16:24:36.586674   999 camera_fxtof1.cpp:233] Found firmware for mode: near
I0527 16:24:36.586736   999 camera_fxtof1.cpp:236] Firmware size: 14404 bytes
I0527 16:24:38.400131   999 calibration_fxtof1.cpp:278] Camera intrinsic parameters:
    fx: 370.694
    fy: 370.57
    cx: 318.164
    cy: 235.115
W0527 16:25:09.442328  1005 addi9036_sensor.cpp:845] select timeout
W0527 16:25:09.443333  1005 camera_fxtof1.cpp:371] Failed to get frame from device

```

SeptimiuVana, 2021-05-28:
Hello, 

Please make sure that the interposer is connected to an external power supply, the problem might be that RPi doesn't provide enough power through it's USB's.

KamelAbdelouahab, 2021-05-28:
Thanks for your feedback,

The interposer was already connected to a 5V/2.4A external power supply. 
I can user higher power delivery if needed.

SeptimiuVana, 2021-05-31:
Hello, 
Can you please try with another power supply and and another USB cable?

KamelAbdelouahab, 2021-06-03:
Unfortunately, I'm facing the same issues even after changing the power supply and USB cable.

KamelAbdelouahab, 2021-06-10:
It seems like I had a hardware failure on my camera. 
On a new device, I'm able to capture frames normally. 
Thanks for your assistance,
Kamel

closed 2021-06-10 by KamelAbdelouahab

mathklk, 2023-05-02:
Switching to a seperate power supply did the job for me, i had the interposer connected to the raspi. The [user guide](https://wiki.analog.com/resources/eval/user-guides/ad-fxtof1-ebz/ug_rpi) does say to use a seperate supply, but I feel like it should explicitly say that you cannot use the raspi's usb port.