analogdevicesinc/aditof_sdk #563: where can get 96TOF functions documents in python

SumingNTUT, 2021-05-28:
Hi 
If I want to set camera status to idle, close or when I trigger camera that grab IR or depth map
How can I do ?

thanks

dNechita, 2021-05-31:
Hi,
You want to stop the camera from streaming? And then start it again? And so on?

SumingNTUT, 2021-05-31:
Yes, I want set "stop" or "start" to camera, and does camera have trigger model from software or hardware?

SumingNTUT, 2021-05-31:
> Hi,
> You want to stop the camera from streaming? And then start it again? And so on?
Yes, I want set "stop" or "start" to camera, and does camera have trigger model from software or hardware?

dNechita, 2021-06-02:
Hi,

Currently, we don't have documentation for the Python API. All available functions and types can be found here: https://github.com/analogdevicesinc/aditof_sdk/blob/master/bindings/python/aditofpython.cpp
You can take a look at some of our examples to see how the Python calls are being made: https://github.com/analogdevicesinc/aditof_sdk/blob/master/bindings/python/examples/

The start and stop don't work for this camera due to a driver limitation. The camera will start streaming when calling the setFrameType() method of the Camera class. And it will stop streaming once the Camera instance is destroyed.

The camera trigger is hardware.

SumingNTUT, 2021-06-02:
> hardware

closed 2021-06-02 by SumingNTUT