analogdevicesinc/aditof_sdk #567: XavierNX device tree and cam-i2c-mux

KamelAbdelouahab, 2021-07-02:
Hi all,

We are working with an FX-TOF camera and a XavierNX, and we would like to attach a second (non ToF) camera on the second CSI port on the devkit. 

The cam-i2cmux part is giving us troubles as the two only available entries are occupied by the ADDI tof camera. Knowing that Nvidia uses a 2->1 i2c mux in their devkit, we managed to successfully register the two cameras by bypassing the mux for camera 2 (we can provide the whole dts if needed): 

```
cam_i2cmux {
		compatible = "i2c-mux-gpio";
		#address-cells = <1>;
		#size-cells = <0>;
		i2c-parent = <&cam_i2c>;
		mux-gpios = <&tegra_aon_gpio CAM_I2C_MUX GPIO_ACTIVE_HIGH>;
		i2c@0 {
               // ADI ToF
		};
		i2c@1 {
               // ADI IR
		};
};
i2c@3180000 {
		#address-cells = <1>;
		#size-cells = <0>;	
		status = "okay";
                // Camera 2 scope ...
};
```

However, it's impossible to capture frames from both devices simultaneously as one device occupies the hole bus during capture. FX-TOX addresses are not acknowledged  when using camera 2 and vice versa ...

Is it possible to implement the ADDI with only one i2c entry? Otherwise, can we use other I2Cs in a way to keep one mux entry for other cameras ? 

Thanks for your assistance,

KamelAbdelouahab, 2021-07-06:
Update: 
We placed ADI_IR on another i2c (the one that goes to the GPIO expander). It seems to work fine.

closed 2021-07-06 by KamelAbdelouahab

dNechita, 2021-07-06:
Thanks for sharing the solution. Glad to hear that things worked out.