analogdevicesinc/aditof_sdk #569: nalyze sensor data from ADFXTOF1-EBZ using Matlab

Dustin-cell, 2021-07-23:
Hi there,

I want to use Matlab on a Windows computer to analyze the sensor data from the ADFXTOF1-EBZ. As a bridge I use the Raspberry Pi 3+. The communication between the Raspberry Pi and the Windows computer should run via ethernet, like in the hardware setup in the following picture.  

[image]

I downloaded the adapter "aditof-adapter-windows-x64.zip" from https://github.com/analogdevicesinc/aditof_sdk/releases/tag/v2.0.0 and unziped the two files into the used Matlab directory.

I think the settings for the Raspberry Pi are described in here: https://github.com/analogdevicesinc/aditof_sdk/tree/master/bindings/matlab. I runed "aditof-server.sh" on the Raspberry Pi and followed this step "-DWITH_MATLAB=on" while building the SDK.

Unfortunately it didn't work so that Matlab prints a connection error while bashing the in this picture described code.

[image]

Does somebody have a more exact instruction or can explain the steps i made wrong or forgot in detail?

That would be great.

Regards Dustin

dNechita, 2021-07-23:
Does the aditof-demo work? (via network)

Dustin-cell, 2021-07-23:
No, the demo does not run via network. The camera is not found.

dNechita, 2021-07-23:
Looks like the demo isn't able to connect to the Raspberry Pi and it complains about the IP address.
Can you ping the IP address of the RPI? Does it reply?

Dustin-cell, 2021-07-26:
Yes, the ping works (output below). I have tried to run the demo again, but the output is the same.

[image]

acozma, 2021-07-26:
A few things to look at:
- is this the SD image that was used for RPi3? https://swdownloads.analog.com/cse/aditof/raspberrypi-fxtof1-latest-image.tar.xz
- can you please try to enable the network serve on RPi, in case it's not already running? https://github.com/analogdevicesinc/aditof_sdk/tree/master/utils/raspberrypi

Dustin-cell, 2021-07-26:
I have checked both points, but there is no change in the output of the demo terminal.

[image]

SeptimiuVana, 2021-07-28:
Hello, we are investigating the problem and we will come back with an answer soon.

Hakunadi, 2021-08-06:
Is there an update for this?

SeptimiuVana, 2021-08-10:
Hello, the solution to your problem can be fixed by following this steps:
1. Checkout at the latest master on both rpi and windows.
2. when building the project please add -DUSE_FXTOF1=1 alongside with -DWITH_MATLAB=on on windows and -DUSE_FXTOF1=1 -DRASPBERRYPI=on on rpi
3. run the server on pi manually to see the status of the connection, it is located in aditof_sdk/build/apps/server/aditof-server. 
(if a port error appears it can be fixed following the steps from here: https://github.com/analogdevicesinc/aditof_sdk/tree/master/apps/server )
4. If the server runs please try aditof-demo first (to check the connection) before runing the matlab example.

Dustin-cell, 2021-08-10:
Regarding the network server and the port error down below in the screenshot. Where do I find out what has to be entered as port value instead of 5000?

[image]

SeptimiuVana, 2021-08-11:
https://github.com/analogdevicesinc/aditof_sdk/blob/master/apps/server/server.cpp#L245
https://github.com/analogdevicesinc/aditof_sdk/blob/master/sdk/src/connections/network/network.cpp#L152

The value can be a random value between 1023 and 65353 but it has to be the same in both locations.

Dustin-cell, 2021-08-11:
Ok, that worked out fine, the error message does not occur anymore. 

Now it is probably due to the building of the SDK on Windows. So far I have not been able to build it. 
First of all, where can I download it? I think it should be the following download: https://github.com/analogdevicesinc/aditof_sdk/archive/refs/tags/v2.0.0.zip 

Next, like mentioned in "Building the SDK in Visual Studio using the script" (https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/windows/build_instructions.md) I have installed OpenSSL and OpenCV. Then I followed the steps described, however no adi_tof_project.sln-file to open in Visual Studio was generated. 

So what would be the concrete steps to build my application at this point? Maybe there is a more detailed explanation  .
Or is it possible to build the SDK on windows via a simpler way?

SeptimiuVana, 2021-08-12:
Hello, if you want to use the script please run it via powershell (administrator version). If the adi_tof_project.sln file is not generated please send us the log.

Dustin-cell, 2021-08-12:
Dear SeptimiuVana,

I am afraid, but I can not follow your desultory instructions, so our conversation in this forum does not come to anything. 

Nevertheless, to make things easy. 

Would it be possible for you just to make a clear tutorial including all the steps to be done so that I can end up using the recorded sensor data in Matlab ? 
Similar to your post from 10/08/2021, but with more detail, pleace.
Ideally with all the links, programs and commands needed for the Windows computer and the Raspberry Pi. 

Would be great. 
Thank you very much in advance.

acozma, 2021-08-16:
The easiest way to build the SDK on Windows is to use the build script for Visual Studio. Most probably the Visual Studio solution file was not generated because of an error in the build process. Can you please post the output of the setup_project.bat?

The most common errors in the build process are:
- the OPENSSL_ROOT_DIR system variable is missing and must be defined in the System Environmental variables - e.g. OPENSSL_ROOT_DIR=C:\OpenSSL-Win64
- OpenCV was not added to the system path at installation time

To generate also the MATLAB adapter [this line](https://github.com/analogdevicesinc/aditof_sdk/blob/master/scripts/windows/setup_project.bat#L217) in the build script must be changed to: set CMAKE_OPTIONS=-DWITH_MATLAB=on

To get the fix for the MATLAB connectivity you'll have to clone the master branch of the sdk using git: git clone https://github.com/analogdevicesinc/aditof_sdk

Dustin-cell, 2021-08-17:
This is the output of the setup_project.bat.

[image]

acozma, 2021-08-17:
It looks like git is not installed or not added to the system path so that it can be called from any location. The git installer can be found here: https://git-scm.com/downloads

Dustin-cell, 2021-08-17:
Git was not installed yet. Now it is the following output. The adi_tof_project.sln-file is still  not created.

`PS C:\Users\trabi> C:\Users\ \aditof_sdk-2.0.0\scripts\windows\setup_project.bat
Setup will continue with the configuration: Release
Setup will continue with the generator: "Visual Studio 15 2017 Win64"
The sdk will be built in: C:\Users\ \build
The deps will be downloaded in: C:\Users\ \deps
The deps will be installed in:  C:\Users\ \deps\installed
Do you want to continue? [Y/N]Y
Please enter OpenSSL root absolute path: OPENSSL_ROOT_DIR=C:\Users\ \SDK_build\OpenSSL\OpenSSL-Win64
OPENSSL_ROOT_DIR=C:\Users\ \SDK_build\OpenSSL\OpenSSL-Win64
Please enter opencv 3.4.1 root absolute path: C:\Users\ \SDK_build\OpenCV\opencv
C:\Users\ \SDK_build\OpenCV\opencv
"Installing glog with config=Release and generator="Visual Studio 15 2017 Win64""
Cloning into 'glog'...
remote: Enumerating objects: 137, done.
remote: Counting objects: 100% (137/137), done.
Receiving objects:  10% (14/137)% (123/123), done. eceiving objects:   9% (13/137)
remote: Total 137 (delta 17), reused 59 (delta 6), pack-reused 0
Receiving objects: 100% (137/137), 570.78 KiB | 1.40 MiB/s, done.
Resolving deltas: 100% (17/17), done.
Note: switching to 'a6a166db069520dbbd653c97c2e5b12e08a8bb26'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at a6a166d Update the version in CMakeLists.txt to 0.3.5
CMake Error at CMakeLists.txt:11 (project):
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

-- Configuring incomplete, errors occurred!
See also "C:/Users/ /deps/glog/build_0_3_5/CMakeFiles/CMakeOutput.log".
CMake Error:
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

"Installing protobuf with config=Release and generator="Visual Studio 15 2017 Win64""
Cloning into 'protobuf'...
remote: Enumerating objects: 2409, done.
remote: Counting objects: 100% (2409/2409), done.
remote: Compressing objects: 100% (1742/1742), done.
remote: Total 2409 (delta 930), reused 1303 (delta 544), pack-reused 0
Receiving objects: 100% (2409/2409), 5.16 MiB | 1.12 MiB/s, done.
Resolving deltas: 100% (930/930), done.
Note: switching to '6a59a2ad1f61d9696092f79b6d74368b4d7970a3'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Updating files: 100% (2206/2206), done.
CMake Error at CMakeLists.txt:20 (project):
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

-- Configuring incomplete, errors occurred!
See also "C:/Users/ /deps/protobuf/build_3_9_0/CMakeFiles/CMakeOutput.log".
CMake Error:
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

"Installing websockets with config=Release and generator="Visual Studio 15 2017 Win64""
Cloning into 'libwebsockets'...
remote: Enumerating objects: 863, done.
remote: Counting objects: 100% (863/863), done.
remote: Compressing objects: 100% (728/728), done.
remote: Total 863 (delta 174), reused 437 (delta 109), pack-reused 0
Receiving objects: 100% (863/863), 5.21 MiB | 1.26 MiB/s, done.
Resolving deltas: 100% (174/174), done.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error at CMakeLists.txt:180 (project):
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

-- Configuring incomplete, errors occurred!
See also "C:/Users/ /deps/libwebsockets/build_3_1_stable/CMakeFiles/CMakeOutput.log".
CMake Error:
  Generator

    Visual Studio 15 2017 Win64

  could not find any instance of Visual Studio.

CMake Error: The source directory "C:/" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
Error: could not load cache
Error: could not load cache`

acozma, 2021-08-17:
The problem is caused by this error message: "could not find any instance of Visual Studio" - to compile on Windows Visual Studio needs to be installed. 

By default, the script is using Visual Studio 15 2017, if there is another version of VS installed it can be specified as a parameter to the script like this: .\setup_project.bat -g "Visual Studio 14 2015 Win64" -c Release

The following configurations are supported: 
- Visual Studio 16 2019
- Visual Studio 15 2017 Win64
- Visual Studio 14 2015 Win64

Dustin-cell, 2021-08-17:
I have the Microsoft Visual Studio Community 2019 Version 16.7.4. So I tried running the command ".\setup_project.bat -g "Visual Studio 16 2019" -c Release". But there is still the massage "could not find any instance of Visual Studio".

acozma, 2021-08-17:
In this case most probably Visual Studio didn't get properly registered in the system at installation time and CMake does not see it.
One option is to try to reinstall Visual Studio. The other option is to wait until tomorrow and we'll make a pre-release with all the fixes. This way you'll not have to compile the SDK.

Dustin-cell, 2021-08-17:
All right, in that case I will wait until tomorrow.

SeptimiuVana, 2021-08-18:
Hello,

Please go to the following link and download aditof-adapter-windows.zip and aditof-setup.exe.
aditof-setup.exe will install the demo application so that we can test the connection and aditof-adapter-windows.zip contains the libraries for the sdk as well as the bindings with matlab.
https://github.com/analogdevicesinc/aditof_sdk/releases/tag/v2.0.1

After you install aditof-setup.exe please power on the raspberrypi and do the following steps:
1.  press `CTRL + ALT + T" to open up a terminal 
2. type: cd workspace/github/aditof_sdk/build
3. type: git pull
4. type: git checkout tags/v2.0.1
5. type: make -j4
6. type: sudo ./examples/aditof-demo/aditof-demo and test the application, it should work.
7.After testing the application please go back to the terminal and
type: sudo ./apps/server/aditof-server to start the server on pi
 
The ip address can be seen in the top right corner if you hover the mouse over the wifi/ethernet conmection
 
 After the server has been started please run the aditof-demo.exe application installed on windows to check the conmexion.

If the connection is stable please copy the files from aditof-adapter-windows.zip (2 libraries) in Documents/MATLAB. 
Open matlab and open the aditof-imaq.m script from aditof_sdk/bindings/matlab/

If you run the script it should work now.

Best regards,
Septimiu

Dustin-cell, 2021-08-18:
The steps just worked out fine until the aditof-demo. 

The ouput in the terminal of the aditof-demo on windows is still the same as before. Now the aditof-server does not show anything about the connection. With the old SDK there was an error message.

Matlab runs the aditof-imaq.m.  Also in that case the aditof-server does not show anything about the connection.  Figure 1, which should display the image in the near mode shows just whole red picture. Figure 2 (Medium Mode) a whole black image. Meanwhile the sensor does not draw any current.

acozma, 2021-08-18:
Just to make sure, did step 6 work? This validates the ToF camera works on the Raspberry Pi.

If step 6 didn't work there might be a few reasons:
 - the flex cables between the ToF module and the interposer, and between the interposer and the RPi are not properly connected
 - in the AD-FXTOF1-EBZ box there were two 15 pins flex cables, one with contacts on the same side and one with contacts on the opposite sides. The cable with contacts on opposite sides must be used.
 - the power supply for the ToF module cannot supply enough current - it is recommended to use a supply with at least 2A current source capability

Dustin-cell, 2021-08-19:
Yes, on the Raspberry Pi the demo application still works fine.

SeptimiuVana, 2021-08-20:
Hello, we are preparing a new raspberry pi image to make sure everything is up-to-date and we will come back with an answer soon.

SeptimiuVana, 2021-08-23:
Hello,

Please download the latest raspberrypi image from the following link:
https://swdownloads.analog.com/cse/aditof/raspberrypi-fxtof1-latest-image.tar.xz

And download the aditof-demo installer and matlab-adapter again from the following link:
https://github.com/analogdevicesinc/aditof_sdk/releases/tag/v2.0.1

Dustin-cell, 2021-08-24:
After the updates, both the demo application on Windows and the aditof_imaq.m work. Thanks for that.

As soon as I start my own program with the functions from the aditof_imaq.m and there is an error message, the application no longer works over the network and I have to rebuild the Raspberry Pi image. Restarting has no effect.

What could go wrong in Matlab so that the mentioned error occurs? Possibly an unintentional change of a file on the Raspberry PI?

SeptimiuVana, 2021-08-25:
Hello, when you restart do you poweroff the camera as well or only the rpi? Can you send the log from pi and windows as well?

Hakunadi, 2021-09-30:
You can try to setup a manual server connection ---   ** .sudo ./aditof.server **

closed 2021-10-29 by acozma