analogdevicesinc/aditof_sdk #571: Python scripts not working on Raspberry Pi 3

acozma, 2021-07-26:
We are having some trouble getting the Python scripts working. The Demo itself works but when we try to run the example python scripts we get an Import Error concerning ‘import aditofpython as tof’ the error reads:

‘ImportError: /usr/local/lib/python3.7/dist-packages/aditofpython.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: _ZN6aditof6System10initializeEv’

We think it is related to the fact that we haven’t built the SDK but when we follow the build instructions in the /doc/raspberrypi3/ folder and enter the commands we get ‘no such file or directory’ for each of the commands even though we know the directory is there. 

Have we missed a step when setting the Raspberry Pi up?

SeptimiuVana, 2021-07-27:
There are 2 solutions to this problem:
1) install the project from the build folder with: `sudo cmake --build . --target install`  and run the script after.
2) copy `aditofpython.cpython-37m-arm-linux-gnueabihf.so` from `aditof_sdk/build/bindings/python` next to the script that you want to run.

Louie2411, 2021-07-27:
> There are 2 solutions to this problem:
> 
> 1. install the project from the build folder with: `sudo cmake --build . --target install`  and run the script after.
> 2. copy `aditofpython.cpython-37m-arm-linux-gnueabihf.so` from `aditof_sdk/build/bindings/python` next to the script that you want to run.

Using the second solution has fixed this error, thank you! However, I am now met with a new error when trying to run the example scripts 'dnn.py' and 'first_frame.py'. This error reads: 
`status = system.initialize()`
`AttributeError: 'aditofpython.System' object has no attribute 'initialize'`

SeptimiuVana, 2021-07-27:
Hello, there might be a version error. Please delete the build folder, checkout at the latest master branch on git and build the project again. 
Create a new build folder and type in the terminal inside the folder:
`cmake -DWITH_PYTHON=on -DWITH_EXAMPLES=on -DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" ..`
`cmake --build . --target install"
for AD-96TOF1-EBZ or
`cmake -DUSE_FXTOF1=1 -DWITH_PYTHON=on -DWITH_EXAMPLES=on -DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" ..`
`cmake --build . --target install"
for AD-FXTOF1-EBZ.

PS: please use sudo python3 to run the script.

Louie2411, 2021-07-27:
Hi, I am trying to follow the instructions you have provided, however I get 'Error: Could not load cache. I have provided the console window of what I have done. 
[image]

SeptimiuVana, 2021-07-27:
Hello, please run the provided cmake commands from ~/Downloads/aditof_sdk-master/build instead. (The folder must be created)

Louie2411, 2021-07-27:
Hi, I made a new build folder but I think it had an error again, I tried running the dnn.py script again and had the same error: 
[image]

closed 2021-11-15 by dNechita