analogdevicesinc/aditof_sdk #575: ROS Node using the AD-FXTOF1-EBZ is not running properly.

analog-cprende, 2021-08-11:
I am trying to get the aditof_sdk ROS node running with the AD-FXTOF1-EBZ on a Raspberry Pi 4. While I have resolved the build issues, I am getting errors when I launch the node using the launch file (roslaunch aditof_roscpp camera_node.launch) as follows:
```
roslaunch aditof_roscpp camera_node.launch
... logging to /home/pi/.ros/log/fa3affde-facc-11eb-9eac-dca63276f009/roslaunch-raspberrypi-4015.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.68.115.72:41327/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    aditof_camera_node (aditof_roscpp/aditof_camera_node)
    reconfigParams (rqt_reconfigure/rqt_reconfigure)

auto-starting new master
process[master]: started with pid [4025]
ROS_MASTER_URI=http://10.68.115.72:11311

setting /run_id to fa3affde-facc-11eb-9eac-dca63276f009
process[rosout-1]: started with pid [4036]
started core service [/rosout]
process[aditof_camera_node-2]: started with pid [4039]
process[reconfigParams-3]: started with pid [4041]
I0811 20:53:07.038094  4049 aditof_utils.cpp:51] No ip provided, attempting to connect to the camera through USB
I0811 20:53:07.039536  4049 sensor_enumerator_raspberrypi.cpp:49] Looking for sensors on the target: Raspberry PI
I0811 20:53:07.041591  4049 camera_fxtof1.cpp:127] Initializing camera
I0811 20:53:07.041821  4049 addi9036_sensor.cpp:140] Opening device
I0811 20:53:07.042023  4049 addi9036_sensor.cpp:158] Looking for the following cards:
I0811 20:53:07.042098  4049 addi9036_sensor.cpp:160] unicam
E0811 20:53:07.051681  4049 tmp10x_sensor.cpp:60] Temp sensor open error
I0811 20:53:07.876914  4049 camera_fxtof1.cpp:178] Camera initialized
I0811 20:53:07.879365  4049 addi9036_sensor.cpp:257] Starting device 0
I0811 20:53:07.881724  4049 camera_fxtof1.cpp:209] Chosen mode: near
I0811 20:53:07.881794  4049 camera_fxtof1.cpp:222] Camera range for mode: near is: 250 mm and 800 mm
I0811 20:53:07.881922  4049 camera_fxtof1.cpp:233] Found firmware for mode: near
I0811 20:53:07.881953  4049 camera_fxtof1.cpp:236] Firmware size: 14404 bytes
libEGL warning: DRI2: failed to authenticate
I0811 20:53:09.665556  4049 calibration_fxtof1.cpp:290] Camera intrinsic parameters:
    fx: 323.025
    fy: 0
    cx: 374.073
    cy: 0
I0811 20:53:09.755007  4049 camera_fxtof1.cpp:209] Chosen mode: medium
I0811 20:53:09.755093  4049 camera_fxtof1.cpp:222] Camera range for mode: medium is: 300 mm and 3000 mm
I0811 20:53:09.755262  4049 calibration_fxtof1.cpp:290] Camera intrinsic parameters:
    fx: 323.025
    fy: 0
    cx: 374.073
    cy: 0
W0811 20:53:09.790287  4049 camera_fxtof1.cpp:439] Unsupported control
E0811 20:53:09.790509  4049 aditof_utils.cpp:137] Could not set camera revision!
[aditof_camera_node-2] process has died [pid 4039, exit code -7, cmd sudo /home/pi/Downloads/aditof_sdk/build/catkin_ws/devel/lib/aditof_roscpp/aditof_camera_node  __name:=aditof_camera_node __log:=/home/pi/.ros/log/fa3affde-facc-11eb-9eac-dca63276f009/aditof_camera_node-2.log].
log file: /home/pi/.ros/log/fa3affde-facc-11eb-9eac-dca63276f009/aditof_camera_node-2*.log

```
Note: I have modified the launch file to give the camera node sudo priviliges as follows:
```
<launch>
  <arg name="ip" default='""'/>
  <node pkg="aditof_roscpp" type="aditof_camera_node" name="aditof_camera_node" launch-prefix="sudo" args="$(arg ip)" output="screen"/>
  <node pkg="rqt_reconfigure" type="rqt_reconfigure" name="reconfigParams"/>
</launch>

```
This allows the node to access the onboard EEPROM without errors.

I am following the Raspberry Pi4 setup instructions here: https://wiki.analog.com/resources/eval/user-guides/ad-fxtof1-ebz/ug_rpi
I use the ADI SD Card image and have verified that the Desktop aditof_demo.sh program works with the sensor ( although it does not allow me to enter a file name to record data, there is an issue with entering characters in that field ).

In order to setup the SDK/ROS node I follow the instructions here: https://github.com/analogdevicesinc/aditof_sdk/blob/master/doc/raspberrypi3/build_instructions.md 
and here: https://github.com/analogdevicesinc/aditof_sdk/tree/master/bindings/ros

Since ROS-Melodic is not available as an apt package I need to build from source. I follow these instructions for this:
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Melodic%20on%20the%20Raspberry%20Pi
In step 3.1, I install the desktop verion and no additional packages are installed. There were no errors encountered doing this.

Here is a set of the make and build commands that I run to build the ROS node and examples for the FXTOF1-EBZ.

```
cmake -DWITH_EXAMPLES=on -DWITH_ROS=on -DWITH_NETWORK=on -DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" -DRASPBERRYPI=1 -DUSE_FXTOF1=1 ..
sudo cmake --build . --target install
cmake --build . --target aditof_ros_package
```
Since I build the examples I can verify that the aditof-demo program works correctly. However it needs sudo permissions, otherwise I get the following errors:
```
./aditof-demo 
I0811 21:02:26.007210  4227 sensor_enumerator_raspberrypi.cpp:49] Looking for sensors on the target: Raspberry PI
I0811 21:02:26.008947  4227 camera_fxtof1.cpp:127] Initializing camera
I0811 21:02:26.009008  4227 addi9036_sensor.cpp:140] Opening device
I0811 21:02:26.009061  4227 addi9036_sensor.cpp:158] Looking for the following cards:
I0811 21:02:26.009089  4227 addi9036_sensor.cpp:160] unicam
E0811 21:02:26.017079  4227 eeprom.cpp:64] fopen() failed. Error: Permission denied
E0811 21:02:26.017431  4227 camera_fxtof1.cpp:156] Failed to open EEPROM with name 24c1024
I0811 21:02:26.019940  4227 addi9036_sensor.cpp:257] Starting device 0
I0811 21:02:26.052780  4227 camera_fxtof1.cpp:209] Chosen mode: near
I0811 21:02:26.052925  4227 camera_fxtof1.cpp:222] Camera range for mode: near is: 250 mm and 800 mm
I0811 21:02:26.053064  4227 camera_fxtof1.cpp:233] Found firmware for mode: near
I0811 21:02:26.053088  4227 camera_fxtof1.cpp:236] Firmware size: 14404 bytes
W0811 21:02:27.871995  4227 calibration_fxtof1.cpp:238] Invalid intrinsic 5.000000
W0811 21:02:27.872748  4227 calibration_fxtof1.cpp:288] Failed to read intrinsic from eeprom

```
The program still starts and I can play the depth and ir data live.
Any help with this would be appreciated.
Thank you,
Colm

analog-cprende, 2021-08-12:
Just a quick update. I am running into a similar issue with the ROS node and the AD-96TOF1-EBZ board on the same Raspberry Pi. As per the instructions I am running the follow commands:

```
cmake -DWITH_EXAMPLES=on -DWITH_ROS=on -DWITH_NETWORK=on -DCMAKE_PREFIX_PATH="/opt/glog;/opt/protobuf;/opt/websockets" -DRASPBERRYPI=1 ..
sudo cmake --build . --target install
cmake --build . --target aditof_ros_package
source ../setup_ros1_local 
roslaunch aditof_roscpp camera_node.launch

```
The ROS node fails with the follow output:

```
$ roslaunch aditof_roscpp camera_node.launch
... logging to /home/pi/.ros/log/ec8c5f5e-fb7b-11eb-bed7-dca63276f009/roslaunch-raspberrypi-4682.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.68.115.72:33115/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    aditof_camera_node (aditof_roscpp/aditof_camera_node)
    reconfigParams (rqt_reconfigure/rqt_reconfigure)

auto-starting new master
process[master]: started with pid [4692]
ROS_MASTER_URI=http://10.68.115.72:11311

setting /run_id to ec8c5f5e-fb7b-11eb-bed7-dca63276f009
process[rosout-1]: started with pid [4703]
started core service [/rosout]
process[aditof_camera_node-2]: started with pid [4706]
process[reconfigParams-3]: started with pid [4708]
I0812 17:45:25.952389  4716 aditof_utils.cpp:51] No ip provided, attempting to connect to the camera through USB
I0812 17:45:25.953450  4716 sensor_enumerator_raspberrypi.cpp:49] Looking for sensors on the target: Raspberry PI
I0812 17:45:25.955127  4716 camera_96tof1.cpp:146] Initializing camera
I0812 17:45:25.955212  4716 addi9036_sensor.cpp:140] Opening device
I0812 17:45:25.955287  4716 addi9036_sensor.cpp:158] Looking for the following cards:
I0812 17:45:25.955318  4716 addi9036_sensor.cpp:160] unicam
I0812 17:45:26.346835  4716 calibration_96tof1.cpp:162] EEPROM calibration data size 64940 bytes
libEGL warning: DRI2: failed to authenticate
I0812 17:45:32.147253  4716 camera_96tof1.cpp:202] Camera initialized
I0812 17:45:32.149843  4716 addi9036_sensor.cpp:257] Starting device 0
I0812 17:45:32.152070  4716 camera_96tof1.cpp:233] Chosen mode: near
I0812 17:45:32.152118  4716 camera_96tof1.cpp:277] Camera range for mode: near is: 250 mm and 800 mm
I0812 17:45:32.152806  4716 camera_96tof1.cpp:287] Found firmware for mode: near
I0812 17:45:32.152851  4716 camera_96tof1.cpp:290] Firmware size: 10440 bytes
I0812 17:45:33.459466  4716 calibration_96tof1.cpp:376] Camera calibration parameters for mode: near are gain: 1 offset: 0
I0812 17:45:33.459908  4716 calibration_96tof1.cpp:388] Camera intrinsic parameters:
    fx: 391.067
    fy: 386.995
    cx: 347.377
    cy: 233.733
I0812 17:45:33.594604  4716 camera_96tof1.cpp:233] Chosen mode: medium
I0812 17:45:33.594679  4716 camera_96tof1.cpp:277] Camera range for mode: medium is: 300 mm and 4500 mm
I0812 17:45:33.595304  4716 camera_96tof1.cpp:287] Found firmware for mode: medium
I0812 17:45:33.595335  4716 camera_96tof1.cpp:290] Firmware size: 10388 bytes
I0812 17:45:34.881682  4716 calibration_96tof1.cpp:376] Camera calibration parameters for mode: medium are gain: 1.5 offset: 0
I0812 17:45:34.881901  4716 calibration_96tof1.cpp:388] Camera intrinsic parameters:
    fx: 391.067
    fy: 386.995
    cx: 347.377
    cy: 233.733
[aditof_camera_node-2] process has died [pid 4706, exit code -7, cmd sudo /home/pi/Downloads/aditof_sdk/build/catkin_ws/devel/lib/aditof_roscpp/aditof_camera_node  __name:=aditof_camera_node __log:=/home/pi/.ros/log/ec8c5f5e-fb7b-11eb-bed7-dca63276f009/aditof_camera_node-2.log].
log file: /home/pi/.ros/log/ec8c5f5e-fb7b-11eb-bed7-dca63276f009/aditof_camera_node-2*.log
```
Thank you.
Colm.

acozma, 2021-08-16:
Can you please post the crash log?

analog-cprende, 2021-08-17:
Hi Andre,
Attached is a copy of the ROS log directory associated with this error and the roslaunch terminal log.
After reading the camera intrinsics twice the node outputs these messages before crashing:
```
W0817 16:56:47.391522  6351 camera_fxtof1.cpp:440] Unsupported control
E0817 16:56:47.391817  6351 aditof_utils.cpp:137] Could not set camera revision!
```
Thank you !

Attached files below:
[adtof_ros_error_log.zip](https://github.com/analogdevicesinc/aditof_sdk/files/7000501/adtof_ros_error_log.zip)
[roslaunch_term.log](https://github.com/analogdevicesinc/aditof_sdk/files/7000522/roslaunch_term.log)