analogdevicesinc/aditof_sdk #683: Review ROS bindings

dNechita, 2021-11-15:
Get feedback

rbudai98, 2021-11-15:
Suggestions from university:

- :depthImage_msg.cpp 
change:
dataToRGBA8(*min_range, getRangeMax(camera), frameData);
to:
memcpy(msg.data.data(), frameData, 2 * msg.width * msg.height);

change:
sensor_msgs::image_encodings::RGBA8
to:
sensor_msgs::image_encodings::MONO16
also: MONO16 in message_factory.cpp, DepthImageMsg.

rbudai98, 2021-11-15:
Existing error:
`E1111 12:34:28.678841 3981 aditof_utils.cpp:129] Requested mode is not available
[ WARN] [1636626977.559729824]: Reconfigure callback failed with exception map::at:
[ WARN] [1636626980.188168017]: Reconfigure callback failed with exception map::at:`

closed 2021-11-18 by rbudai98