analogdevicesinc/aditof_sdk #698: Add96tof1: make setFrameType available on target and network

rbudai98, 2021-11-16:


dNechita, 2021-11-17:
Well, setFrameType function is available on target and on network. It has been for a long time.
This is where setFrameType is implemented in Camera96TOF1:
https://github.com/analogdevicesinc/aditof_sdk/blob/master/sdk/src/cameras/ad-96tof1-ebz/camera_96tof1.cpp#L366
And this is the Network implementation:
https://github.com/analogdevicesinc/aditof_sdk/blob/master/sdk/src/connections/network/network_depth_sensor.cpp#L280
And this is the Target implementation:
https://github.com/analogdevicesinc/aditof_sdk/blob/master/sdk/src/connections/target/addi9036_sensor.cpp#L355

So the title of this issue should be changed as it creates confusion.

rbudai98, 2021-11-17:
Done

closed 2021-11-22 by rbudai98